PaulStoffregen/OneWire #17: esp8266 parasitic wrong temperatures

tablatronix, 2016-03-13:
Why would my temperatures be wrong in parasitic mode ?

```
ROM = 28 FF C9 30 C2 15 1 80
  Chip = DS18B20
  Data = 1 A0 1 4B 46 1F FF 1F 10 E6  CRC=E6
  Temperature = 26.00 Celsius, 78.80 Fahrenheit

// parasitic
ROM = 28 FF 9C 40 C2 15 1 6F
  Chip = DS18B20
  Data = 1 50 5 4B 46 1F FF 1F 10 35  CRC=35
  Temperature = 85.00 Celsius, 185.00 Fahrenheit
```

Testato, 2016-03-13:
It's not wrong, 85 is an error code.
Reducing your pull-up resistor will help you.

tablatronix, 2016-03-13:
How do you tell ? Didn't see it n the datasheet.

dropped to 2k3
```
ROM = 28 FF C9 30 C2 15 1 80
  Chip = DS18B20
  Data = 1 18 4 4B 46 1F FF 1F 10 6B  CRC=6B
  Temperature = 65.50 Celsius, 149.90 Fahrenheit
```

I had to use a 1k to get actual temp.

Thanks.

But I am curious why, is it just something the lib doesn't detect? I figure bit read problems would fail crc check?

tablatronix, 2016-03-13:
I think i understand the 85 now, indication of device resets. Interesting.

closed 2016-03-14 by tablatronix

akupiec, 2017-01-25:
In my case it's starting to show correct temperature around ~1.9k - it would be nice to add more specific information about needed rezystor. 
I spend some time looking into few lib's sources why shown temperature is incorrect.