PaulStoffregen/OneWire #18: Bad values reading DS18B20 below 0 degrees

Humancell, 2016-04-12:
Hello, 

I've just installed a 1-Wire probe in a deep freezer, and it's below zero ... but the numbers coming back from the sensor are WAY below zero?  Using your temperature example, it's showing -20.37C ... but that seems WAY too cold for a basic household freezer.

I found this link that seems to indicate the example is not accounting for the switch below 0C?

Froosh, 2016-04-12:
That sounds about right.

'Normal' freezer temp should be around -18C, so a deep freeze intended to store food for longer should be colder.

My fridge/freezer is showing -18.32C at the moment.

PaulStoffregen, 2016-04-12:
Please compare against another known-good sensor.

I'm closing this issue.  Please only open issues here if you have done the work to confirm the problem.  Comparing to another sensor is the bare minimum!

closed 2016-04-12 by PaulStoffregen

Humancell, 2016-04-13:
You are correct, and I apologize.  The other thermometer that was being used appears to "stick" just below 20F ... which is what threw us off.  After checking a mercury thermometer it really is at ~-10F ...

Sorry ... our error.