PaulStoffregen/OneWire #19: Add a begin() method

SukkoPera, 2016-05-08:
To better comply with the [Arduino API Style Guide](https://www.arduino.cc/en/Reference/APIStyleGuide), the OneWire bus pin should not be passed in the constructor, but rather through a new begin() method.

This would also allow to init the library from setup(), after reading the actual pin to be used from EEPROM for instance, making it more flexible.

Of course the interface should be backwards-compatible.

gordonthree, 2016-08-06:
@SukkoPera can you test my changes? I've added the begin method, and think I perserved the old constructor based init as well. https://github.com/gordonthree/OneWire

SukkoPera, 2016-08-07:
Can't test it at the moment, but I have looked at the code and it looks correct to me.

RobTillaart, 2017-07-28:
One remark, 
do you want the begin() method to be called more than once, meaning changing pin runtime? [No]

Think that is the reason they set_pin() got removed as it can lead to problems in communications / program logic.  imho begin() should have a flag to see it was called before [boolean] .

orgua, 2017-07-28:
as discussed in #22 

@RobTillaart : on your remark -that leaves us with the current implementation to set the pin number once at the start: `OneWire oneWire(8);`

SukkoPera, 2017-07-28:
@RobTillaart I don't think that is necessary. Though a corresponding end() method that clears things up would be good to have.

RobTillaart, 2017-07-28:
@orgua 
I think the use case @gordonthree  has mentioned makes perfect sense. It is not the way beginners should use the OW lib, but experts that know what they are doing, might indeed need a set_pin();

The implications as far I can oversee:
- new constructor OneWire(); as proposed
- void set_pin(uint8_t pin);  sets a new value for _pin;
and to make it robust:
- int get_pin();  to read the current value of _pin;
- all reads should check if _pin has been initialized (to prevent incorrect calls as _pin always has some value)
- all writes should check if _pin has been initialized (idem).

So a clean/save implementation would have an impact on the footprint, not complex but still.
Only implementing set_pin() is oppportunistic and would lead to discussions and questions on fora.

orgua, 2017-07-28:
your so called experts just use "new" and "delete" if they like dangerous old  c++ or use smartpointers to store a onewire-object and replace it, including all internal states, when they need to. the handle is one pointer to the current object. so no setPin() or end() needed. this job is already handled by constructors and destructors of c++-objects that get callled automatically.

PaulStoffregen, 2018-03-15:
fixed https://github.com/PaulStoffregen/OneWire/commit/6ad660a9435c36946530b1bc28c892bd4cc582d2

closed 2018-03-15 by PaulStoffregen