PaulStoffregen/OneWire #27: ESP8266 Pin16 not working

ScheintodX, 2017-01-19:
pin16 is not working as 1wire pin with the special esp8266 code. It is working with the generic driver.

This is because pin 16 needs special treatment (other registers).

s. e.g. esp8266 wiring: 

extern int ICACHE_RAM_ATTR __digitalRead(uint8_t pin) {
  pwm_stop_pin(pin);
  if(pin < 16){
    return GPIP(pin);
  } else if(pin == 16){
    return GP16I & 0x01;
  }
  return 0;
}

as taken from here: https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_wiring_digital.c

bneedhamia, 2017-10-21:
Another reason pin 16 won't work: OneWire.h contains, under the ESP8266 code, #define PIN_TO_BITMASK(pin)             (1 << pin). Because 1 is an int (16 bits) and the mask is 32 bits, shifting the 1 by 16 results in 0 rather than 0x10000L. Looking at the Adafruit ESP8266 OneWire version replaces this line with #define PIN_TO_BITMASK(pin)             (1UL << (pin)).

tretyakovsa, 2018-11-28:
ESP8266 Pin14 not working!

Emmanuel-FR, 2018-12-14:
> ESP8266 Pin14 not working!

Hello, I had the same problem and spent hours trying to figure why. I found out that it works if you use version 2.3.0 of the onewire library.
It seems to be a regression, I tried other pins and only one worked (the standard pin of the exemple code).

gnalbandian, 2019-12-10:
> Another reason pin 16 won't work: OneWire.h contains, under the ESP8266 code, #define PIN_TO_BITMASK(pin) (1 << pin). Because 1 is an int (16 bits) and the mask is 32 bits, shifting the 1 by 16 [...]

Saved my day!
This should be fixed!
Thanks!

PaulStoffregen, 2019-12-10:
Can anyone else using ESP8266 confirm this is now fixed?