PaulStoffregen/OneWire #28: ESP32 Support?

forthlightning, 2017-02-01:
It currently works intermittently, and gives the following error at compile time using esp-idf

```
In file included from /Users/maxfieldmorrison/NSR/solsense32/Firmware/solsense_esp32/main/include/DallasTemperature.h:22:0,
                 from /Users/maxfieldmorrison/NSR/solsense32/Firmware/solsense_esp32/main/./DallasTemperature.cpp:6:
/Users/maxfieldmorrison/NSR/solsense32/Firmware/solsense_esp32/main/OneWire/OneWire.h:249:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
 #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
  ^
/Users/maxfieldmorrison/NSR/solsense32/Firmware/solsense_esp32/main/./DallasTemperature.cpp: In member function 'void DallasTemperature::setHighAlarmTemp(const uint8_t*, char)':
/Users/maxfieldmorrison/NSR/solsense32/Firmware/solsense_esp32/main/./DallasTemperature.cpp:611:22: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     else if (celsius < -55) celsius = -55;
                      ^
/Users/maxfieldmorrison/NSR/solsense32/Firmware/solsense_esp32/main/./DallasTemperature.cpp: In member function 'void DallasTemperature::setLowAlarmTemp(const uint8_t*, char)':
/Users/maxfieldmorrison/NSR/solsense32/Firmware/solsense_esp32/main/./DallasTemperature.cpp:631:22: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     else if (celsius < -55) celsius = -55;
```

PaulStoffregen, 2017-02-01:
I do not use ESP32.  I'm depending on the ESP community to resolve these sorts of issues and submit pull requests.  As a general rule, the changes should be confined to the defs in OneWire.h, so no other types of boards are impacted.

juanbanc, 2017-05-21:
Are you thinking to add support for ESP32 board? If the answer is Yes, when you think it will be supported?
Thanks for you answer.

PaulStoffregen, 2017-05-23:
I'm depending on the ESP community to resolve these sorts of issues and submit pull requests.

PaulStoffregen, 2017-09-01:
Just merged this.

https://github.com/PaulStoffregen/OneWire/pull/40

Please do not ask *me* for ESP help.  Comment on #40.

stickbreaker, 2018-01-04:
@forthlightning @juanbanc  Here is a working fork pr #47 
It has worked for everyone that has tried it.  [stickbreaker/onewire](https://github.com/stickbreaker/OneWire)

Chuck.

closed 2018-01-05 by forthlightning