PaulStoffregen/OneWire #32: M0 based board compile issue

ssozonoff, 2017-05-13:
Currently trying to use this library with the Seeeduino Lorawan which is based on the Cortex M0 (like the Zero) and defined as __SAMD21G18A__ which I see support for in OneWire.h

However I am getting the following compile error and not sure where to look to fix it.
Any help would be much appreciated. I assume its a type error....

```
Users/ttt/Documents/Arduino/libraries/OneWire/OneWire.cpp: In constructor 'OneWire::OneWire(uint8_t)':
/Users/serge/Documents/Arduino/libraries/OneWire/OneWire.cpp:129:10: error: cannot convert 'volatile PORT_DIR_Type*' to 'volatile uint32_t* {aka volatile long unsigned int*}' in assignment
  baseReg = PIN_TO_BASEREG(pin);
          ^
exit status 1
Error compiling for board Seeeduino LoRaWAN.

```

Serge

PaulStoffregen, 2017-05-15:
Have you contacted Seeed for help?  They seem to have changed the types in their version of the SAMD core library.

ssozonoff, 2017-05-15:
No but I will reach out to them. I did fix it by casting in the OneWire lib but this is obviously a hack. Thanks