PaulStoffregen/OneWire #36: Compile error when using another library

dterweij, 2017-07-17:
**Sketch Compile Error:**
```
C:\Users\Danny\Documents\Arduino\libraries\OneWire\OneWire.cpp:571:42: error: '_crc16_update' was not declared in this scope

         crc = _crc16_update(crc, input[i]);
```

**Using:**
Arduino IDE 1.8.2 (Win10)

**Libraries:**
```
#include <OneWire.h> (https://github.com/PaulStoffregen/OneWire)
#include <DallasTemperature.h> (https://github.com/milesburton/Arduino-Temperature-Control-Library)
#include <VirtualWire.h> (https://github.com/danielesteban/ArduinoLib/tree/master/VirtualWire)
```

**Same issue found here, but not a answer how to resolve:**
[https://forum.arduino.cc/index.php?topic=470991.0](url)

**Goal:**
Send/Receive temperature (and other data in the future) over a 433Mhz transmitter/receiver.

orgua, 2017-07-18:
just go to your oneWire library and change the line 569 from
  `#if defined(__AVR__)`
to 
   `#if 0`

don't know what the problem is, your arduino-software seems to have lost the 
   `#include <util/crc16.h>`
do you use a standard arduino?

PaulStoffregen, 2017-07-18:
Which version of Arduino are you using, and which board is selected in Tools > Boards?  What program are you compiling?

I tried just now with Arduino 1.8.3, Arduino Uno, and the OneWire DS18x20_Temperature example.  It compiles without any error.

You *must* give us all the info needed to reproduce this error.  If none of the examples provided by these libs give the error, you need to post a complete program, where complete means it can be copied into Arduino and the error reproduced by clicking Verify.

Without this info, this issue will be closed....

PaulStoffregen, 2017-07-18:
I tried again just now, using Arduino 1.8.3 and Arduino Uno, with all 3 libraries as mentioned on the forum.  It compiles without any error.

dterweij, 2017-07-19:
Updated to Arduino IDE 1.8.3. I have the same versions.
Boards used: Arduino Uno v3 (Original) and the chinese clone (with ch341 windows driver).
Boardmanager is Arduino/Genuino uno
I have 3 of them currently connected. com port 4, 7 and 3.

I shall do the change mentioned by @orgua

orgua, 2017-07-19:
@dterweij so the problem persists with the software-update? very odd! Maybe some of your self-installed libs in ./documents/arduino/libraries is causing some problems. who knows.
but my bugfix should work because it will switch out this missing routine with a slower routine provided in the onewire-lib-sources.

dterweij, 2017-07-19:
It compiles now @orgua. I am going to finish the sketch in the weekend. I expect it works fine. I am closing 
 the issue.
Thank you all.

closed 2017-07-19 by dterweij

robinsout, 2021-01-28:
Just need to downgrade WineWire library version.
Worked like a charm for me with 2.3.0

MakerAndy1, 2022-02-19:
Just a quick shoutout. Had the exact same problem when using onewire 2.3.6 in VScode with platformIO on Teensy 2.0 (teensy lib 4.15.0) board. When using 2.3.2 of onewire it compiled.