PaulStoffregen/OneWire #37: depower() misses DIRECT_WRITE_LOW(baseReg, bitmask);

RobTillaart, 2017-07-28:
Going through the code of OneWire I noticed that the depower() function does call **DIRECT_MODE_INPUT(baseReg, bitmask);** but does not call **DIRECT_WRITE_LOW(baseReg, bitmask);** which is needed to depower if I understand correctly.

If agreed upon I can make a pull request to fix.  That PR would also refactor the write() and write_bytes() to include  ** if ( !power) depower(); ** to remove duplicate code.

orgua, 2017-07-28:
you don't do a lot of embedded dev, right? writing the pin low is not needed. switching the pin to input- or "high impedance"-mode is enough. the whole implementation of the power-argument is faulty. the bus is powered for a short time even when power==false. i fixed this and several other bugs in my fork: https://github.com/orgua/OneWire

RobTillaart, 2017-07-28:
define a lot :)
If the pin low is not needed, the implementation (in this master) of write() and write_bytes() do not need pin low. At least the code is inconsistent.  
I'll have a look at your fork how you solved it.

orgua, 2017-07-28:
well, for "define a lot": just read the dev history in the main readme.md on the frontpage. 

pin_low has other purposes, especially on AVRs. this current code is a big mixup of ancient half-implemented features. but paul calls it "very mature" and won't change a bit

RobTillaart, 2020-07-05:
I checked my open issues and came across this old one. Not relevant anymore for me so I close it.

closed 2020-07-05 by RobTillaart