PaulStoffregen/OneWire #48: Can this library be used to read the DS1921G thermochron iButton?

jpwhitney, 2018-01-03:
I'd like to be able to start datalogging missions and download recorded temperatures from an iButton such as this (https://www.maximintegrated.com/en/products/digital/data-loggers/DS1921G.html) which uses the OneWire protocol.

This is the datasheet (https://datasheets.maximintegrated.com/en/ds/DS1921G.pdf)

I found this (https://www.maximintegrated.com/en/products/ibutton/software/1wire/wirekit.cfm) development public domain kit for the OneWire devices which has humalog.c (also found here https://github.com/AriZuu/OneWire/blob/master/apps/humalog/humalog.c).

Is this possible?

stickbreaker, 2018-01-04:
@jpwhitney 
Yes you can use this library to access a OneWire compatible Sensor.

You will have to read and understand the data sheet and implement those commands.

There are examples included with this library of how to access a DS18b20,  Read the DS18b20 datasheet. By comparing the DS18b20 examples and the DS18b20 datasheet you should be able to develop a library to access and utilize your sensor.

Chuck.

jpwhitney, 2018-01-04:
@stickbreaker 
Thanks for the recommendation, however it seems that the DS250x_PROM example may be more relevant for my particular case.  I am not interested in accessing the sensor of the DS1921G, which likely is a DS18B20, rather I'm interested in accessing the memory of the device, which has stored times and temperature readings.

After looking through the datasheet I've found a few relevant sections:
1. 

> The hierarchical structure of the 1-Wire protocol is shown in Figure 2. The bus master must first provide one of the seven ROM function commands: Read ROM, Match ROM, Search ROM, Conditional Search ROM, Skip ROM, Overdrive-Skip ROM, or Overdrive-Match ROM. Upon completion of an Overdrive ROM command byte executed at standard speed, the device enters overdrive mode, where all subsequent communication occurs at a higher speed. The protocol required for these ROM function commands is described in Figure 13. After a ROM function command is successfully executed, the memory functions become accessible and the master can provide any one of the seven available commands. The protocol for these memory function commands is described in Figure 10. All data is read and written least significant bit first.

2.
[image]

3. 

> 
> Read Memory [F0h]
> The Read Memory command can be used to read the entire memory. After issuing the command, the master must provide the 2-byte target address. After the 2 bytes, the master reads data beginning from the target address and can continue until the end of memory, at which point logic “0”s are read. It is important to realize that the target address registers contain the address provided. The ending offset/data status byte is unaffected. The hardware of the DS1921G provides a means to accomplish error-free writing to the memory section. To safeguard data in the 1-Wire environment when reading and to simultaneously speed up data transfers, it is recommended to packetize data into data packets of the size of one memory page each. Such a packet would typically store a 16-bit CRC with each page of data to ensure rapid, error-free data transfers that eliminate having to read a page multiple times to verify if the received data is correct (refer to Application Note 114: 1-Wire File Structure for the recommended file structure).

4.
[image]

I'm interested in reading the DATA-LOG MEMORY from ADDRESS 1000h to 17FFh.

The flowchart for that is below:
[image]
10a
[image]
10b

[image]
13a
[image]
13b

Am I on the right track here?

stickbreaker, 2018-01-04:
@jpwhitney yep,
Those flow charts are models to follow for your code. I would implement a 16byte block read to match the 'page' origination of these devices.  This OneWire protocol is very time/delay sensitive. To implement it on the ESP32 PR #47  I had to wrap the bit twiddling sections in 'Critical_Section' functions.  The 240mhz processor spends eons waiting for each bit to flow in or out.  It cannot multitask because of these timing issues.  If you used the ESP32 I would recommend issuing a `yield()` between each 16byt block to allow other task some processor time.
Chuck.

bradbeshaw, 2024-01-22:
Hello,  Have you been able to read the DS1921G temperatures with esp32 ?
I'm still attempting to get this working.