PaulStoffregen/OneWire #49: ESP32 compile error

lefedor, 2018-02-17:
Hello, have build error on esp32, attaching log:

`
Arduino: 1.8.5 (Linux), Board: "Node32s, 80MHz, 921600"

In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:143:0:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h: In function 'uint32_t directRead(uint32_t)':
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:111:13: error: redefinition of 'uint32_t directRead(uint32_t)'
 IO_REG_TYPE directRead(IO_REG_TYPE pin)
             ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.h:60:0,
                 from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:142:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:111:13: note: 'uint32_t directRead(uint32_t)' previously defined here
 IO_REG_TYPE directRead(IO_REG_TYPE pin)
             ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:143:0:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h: In function 'void directWriteLow(uint32_t)':
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:122:6: error: redefinition of 'void directWriteLow(uint32_t)'
 void directWriteLow(IO_REG_TYPE pin)
      ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.h:60:0,
                 from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:142:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:122:6: note: 'void directWriteLow(uint32_t)' previously defined here
 void directWriteLow(IO_REG_TYPE pin)
      ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:143:0:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h: In function 'void directWriteHigh(uint32_t)':
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:131:6: error: redefinition of 'void directWriteHigh(uint32_t)'
 void directWriteHigh(IO_REG_TYPE pin)
      ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.h:60:0,
                 from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:142:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:131:6: note: 'void directWriteHigh(uint32_t)' previously defined here
 void directWriteHigh(IO_REG_TYPE pin)
      ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:143:0:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:140:6: error: redefinition of 'void directModeInput(uint32_t)'
 void directModeInput(IO_REG_TYPE pin)
      ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.h:60:0,
                 from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:142:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:140:6: note: 'void directModeInput(uint32_t)' previously defined here
 void directModeInput(IO_REG_TYPE pin)
      ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:143:0:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:168:6: error: redefinition of 'void directModeOutput(uint32_t)'
 void directModeOutput(IO_REG_TYPE pin)
      ^
In file included from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.h:60:0,
                 from /home/lefedor/Arduino/libraries/OneWire-master/OneWire.cpp:142:
/home/lefedor/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:168:6: note: 'void directModeOutput(uint32_t)' previously defined here
 void directModeOutput(IO_REG_TYPE pin)
      ^
Error compiling for board Node32s.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

`

PaulStoffregen, 2018-02-18:
Perhaps you extracted a new ZIP file without deleting the old folder, resulting in a mixture of new and old code?

lefedor, 2018-02-19:
No, it's not that case i've checked of cource, and rechecked today.
If I delete folder there is no OneWire.h found error.

It compiles fine with this version i've found from last commit:
https://github.com/stickbreaker/OneWire

It not compiles with actual repo for esp32.

lefedor, 2018-02-19:
I see there are two inclusions for the
`#include "util/OneWire_direct_gpio.h"` in both OneWire.h and OneWire.cpp - could be it's re-declarations are the reason of error?

BUBBLEIO, 2018-03-02:
Same issue here with ESP32 (generic card profile) compilation. It seems that the `#def and #undef` "strange game" of OneWire_direct_gpio.h make compiler confused!
This compiler issue appear in commit ebc5f7 'Move direct I/O macros, prevent leaking to Arduino sketches' 

I'm really interested to test it on my ESP32. V2.3.3 is working with a Dallas oneWire sensor but fail if I add a BMP280 on I2C on the board. With another I2C it is OK. I use Adafruit_BMP280 library, I've to test SparkFun_BME280 if behaviour is different.

BUBBLEIO, 2018-03-02:
I prepare a pull request to resolve compilation error. Undef was a bad idea, or why ?, true and false do the job why `TRUE/FALSE` to define ? #pragma once is one `define/ifdef` less...

lefedor, 2018-03-02:
I use bme280 over i2c in current project it works fine together with 18b20  at
https://github.com/milesburton/Arduino-Temperature-Control-Library + 
https://github.com/stickbreaker/OneWire library combo variant.

BUBBLEIO, 2018-03-02:
@lefedor Thank you for the advise
My conflict is (with ESP32 board) between **DALLAS over ONEWIRE** and **BME280 over I2C**. I think @stickbreaker mod are "pull requested" to @PaulStoffregen repo. I prefer follow @PaulStoffregen code which is the arduino library reference and is in phase to correct a wifi ESP32 issue I'm also interested in.

PaulStoffregen, 2018-03-15:
fixed https://github.com/PaulStoffregen/OneWire/commit/9e08f4e43eadae975a321c323b85b8d1ddc7aa4a

closed 2018-03-15 by PaulStoffregen

xjc666-666, 2024-09-13:
Ah,I've also met some problems, my board is duo256m(maybe you've never heard) ,I tried to compile a project on Arduino ,I use onewire.h(library file in arduino) ,but failed ,

c:\Users\18158\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:429:23: error: 'GPIO_INPUT_VAL' was not declared in this scope
  429 |     return ((GPIO_REG(GPIO_INPUT_VAL) & mask) != 0) ? 1 : 0;

Actually I don't get it. Can somebody help me to deal with this problem???

uzi18, 2024-09-14:
@xjc666-666 some more info? IDE version? core version + settings? onewire version?

xjc666-666, 2024-09-17:
> @xjc666-666 some more info? IDE version? core link/version + settings? onewire version?
> 
> duo256m is not ESP32 core

yeah i know duo256m is not ESP32 core, but my my errors are quite similliar to yours. Amm, so my ide version is Arduino IDE 2.3.3.
i tried to run a project which uses DS18B20 temperature sensor to get temperature data(
[temperature_sensor.zip](https://github.com/user-attachments/files/17026400/temperature_sensor.zip)
) . And my board is Duo256 Dev Module(made by china). And of course, it failed. I guess that Onewire , this library doesn't match this board. And i've tried so many other code which is about temperature sensor. However, none of these codes can run properly on Duo256 Dev Module, either the library file reported an error, or the serial port monitor did not have any data after burning successfully .

xjc666-666, 2024-09-17:
> @xjc666-666 some more info? IDE version? core link/version + settings? onewire version?
> 
> duo256m is not ESP32 core

 

> @xjc666-666 some 

> @xjc666-666 some more info? IDE version? core link/version + settings? onewire version?
> 
> duo256m is not ESP32 core

I know that Duo256 Dev Module (a domestic development board in China) is not from ESP32 series. I use Onewire library file when I try to write about obtaining the ambient temperature code with DS18B20 temperature sensor on Arduino IDE2.3.2.(
[temperature_sensor.zip](https://github.com/user-attachments/files/17026560/temperature_sensor.zip)
) However, the error message I mentioned before appeared. I suspect that the Onewire library file is incompatible with the development board of the Duo256 Dev Module. Later, I went to find a lot of codes about using the temperature sensor to obtain the temperature, but none of these codes can work properly on the Duo256 dev Module, or the library file reported an error. Or the serial monitor does not have any data after burning successfully

PaulStoffregen, 2024-09-17:
Maybe you should ask the vendor who sold Duo256 for help?

xjc666-666, 2024-09-17:
> Maybe you should ask the vendor who sold Duo256 for help?

yeah i think you're right. Finally i use Arduino nano.