PaulStoffregen/OneWire #56: ESP8266 WTD reset with OneWire

consolacion, 2018-03-24:
Could there be a code defect in the library that causes a WTD in the ESP8266?
Even a simple sketch like:
```
#include <OneWire.h>
// DS18B20 op pin 8.
OneWire ds(8);

void setup() {
Serial.begin(115200); 
Serial.println("Begin" );
}
void loop() {
}

```
causes a WTD in a Wemos D1 mini. Used version 2.3.4, but also reverted back to 2.3.0 to try. Same issue.
Other code on that Wemos has no problems

PaulStoffregen, 2018-03-24:
I do not use ESP8266.  I'm depending on the ESP community to submit pull requests.

Please do not ask questions here.  Only create issues where when you have found a confirmed problem.  If you are unsure, talk first on the ESP community forums.

closed 2018-03-24 by PaulStoffregen

consolacion, 2018-03-24:
Apologies. I thought I was signaling an issue. didn't know it had to be a confirmed issue

PaulStoffregen, 2018-03-24:
Next time, please read the issue template.  It was the text in the box before you posted this.  It specifically says:

> For any question, even questions directly pertaining to this code, post your question on the forums related to the board you are using.

PaulStoffregen, 2018-03-24:
If you do not follow the issue template and provide the required info, I will almost certainly close the issue without any investigation.