PaulStoffregen/OneWire #57: Error ds18b20 with esp32

smacyas, 2018-04-23:
After using the latest Arduino-esp32 there is a problem with ds18b20

Device is preset on the bus, bur ROM data error

https://github.com/espressif/arduino-esp32/issues/1335

paulvha, 2018-12-27:
I found to have the same issue however ONLY THE FIRST time around after reset/reboot. The ESP32 library is using the fastmicroseconds() as mentioned in expressif link above. To get around that the IRAM_ATTR had to be added. As I use the library on different boards and want to keep one source the following 3 changes can be applied:

in Onewire.h (around line 99)
    // Write a bit. The bus is always left powered at the end, see
    // note in write() about that.
#if defined (ARDUINO_ARCH_ESP32)
void IRAM_ATTR write_bit(uint8_t v);
#else
void write_bit(uint8_t v);
#endif

    // Read a bit.
#if defined (ARDUINO_ARCH_ESP32)
uint8_t IRAM_ATTR read_bit(void);
#else
uint8_t read_bit(void);
#endif

in onewire.cpp (around line 204): 
change
   void OneWire::write_bit(uint8_t v)
to
#if defined(ARDUINO_ARCH_ESP32)
void IRAM_ATTR OneWire::write_bit(uint8_t v)
#else
void OneWire::write_bit(uint8_t v)
#endif

in onewire.cpp (around line 236)
change 
  uint8_t OneWire::read_bit(void) 
to
#if defined(ARDUINO_ARCH_ESP32)
uint8_t IRAM_ATTR OneWire::read_bit(void)
#else
uint8_t OneWire::read_bit(void)
#endif

eos1d3, 2019-07-08:
I can confirm the problem for the first incorrect reading when using ESP32. It has been so long that the bug is still there. And the above fixes the problem.

dlcflv, 2020-05-13:
it works! I lost a day trying to understand why the "OneWireSerch" example does not work (it puts the search function in the setup, therefore it is performed only once after the reset, therefore ESP32 does not execute it)
why not integrate it in the official library?

tysonmatanich, 2022-02-07:
The solution offered in https://github.com/milesburton/Arduino-Temperature-Control-Library/issues/168#issuecomment-763158681 works for me. Tested with 1, 2 and 3 sensors. A bit of a hack having to call `sensors.begin()` twice.

```
sensors.begin();
sensors.begin();
```

Used the same wiring setup as https://github.com/milesburton/Arduino-Temperature-Control-Library/issues/85#issuecomment-936510352 on an ESP32­-PICO­-MINI-­02

blazoncek, 2023-02-01:
@PaulStoffregen could you merge one of the (similar) PRs addressing this issue on (multicore?) ESP32?
I've verified with several instances that using IRAM_ATTR solves any issue that I had using OneWire on ESP32 regarding reading bus data.