PaulStoffregen/OneWire #58: ESP8266 parasitic DS18B20 reset during conversion (85°C)

LeoDJ, 2018-04-23:
### Description
I was running two `DS18B20+PAR` temperature sensors without any problems for the last two years on an ESP-01 module running on 3.3V using a 4.7K pullup resistor.

Now, wanting to develop a new firmware I started the development with new versions of everything (Arduino IDE and libraries), but now, with the exact same hardware setup that worked flawlessly for two years, the sensor always returns 85°C (error code for reset during conversion). I suspect that somehow, I don't know how, the ESP does not leave the data pin powered anymore.

To test this, I recompiled and flashed my old firmware that was running fine for two years and lo and behold, it still returned 85°C. Trying to isolate the issue, I flashed the old binary of the firmware and it returned the correct temperatures again.

To isolate the issue further, I tried numerous versions of the OneWire (2.3 to 2.3.2) and [DallasTemperature](https://github.com/milesburton/Arduino-Temperature-Control-Library) (3.7.3 to 3.7.9) libraries, but to no effect. I am still getting 85°C all the time.

I also tried to compile it using Arduino IDE and PlatformIO, but both yielded the same issue still.
Trying on another ESP8266 (Wemos D1 mini and nodeMCU) did not change anything.

Slowly I am suspecting that the issue lies in the ESP8266-Arduino framework updating something or a change in the compiler, which now breaks keeping the data pin configured as output during the conversion.

A workaround, of course, is to use a lower value pullup resistor so that the sensors take power through that instead of the powered microcontroller pin. (One parasitic sensor requires 1K, two 470 Ohms) But this is not a solution.

I will update this issue in the following days by testing on other platforms (older Arduino IDE, older Arduino-ESP8266 version, different OS) and hooking a oscilloscope up to the data line to see if the issue really lies in voltage sag during conversion.
Also, I want to try out the library fork from [orgua](https://github.com/orgua/OneWire) to see if the issue happens in his library too.

So, to conclude: I can rule out a hardware isse, I think it's not an issue with library versions and I suspect something in the ESP8266-Arduino framework or Arduino IDE (compile options).

I am looking forward to help fix this issue soon. :)

### Steps To Reproduce Problem
Try to read temperatures with an ESP8266 and a parasitic DS18B20 with 4.7K pullup. Using the newest Arduino IDE, ESP8266-Arduino, OneWire and DallasTemperature versions.

### Hardware & Software

**Board:** ESP8266 (any module, tried ESP01, Wemos D1 mini, nodeMCU
No shields
**Arduino IDE version:** 1.8.5
**Version info & package name (from Tools > Boards > Board Manager):** 2.3.4 (but tried all other versions)
**Operating system & version:** Windows 10
**Any other software or hardware?** Dallas DS18B20+PAR connected with a 4.7K pullup

### Arduino Sketch
Happens with every sketch, not limited to this one ([Simple Example](https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Simple/Simple.pde))
```cpp
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

/*
 * The setup function. We only start the sensors here
 */
void setup(void)
{
  // start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature IC Control Library Demo");

  // Start up the library
  sensors.begin();
}

/*
 * Main function, get and show the temperature
 */
void loop(void)
{ 
  // call sensors.requestTemperatures() to issue a global temperature 
  // request to all devices on the bus
  Serial.print("Requesting temperatures...");
  sensors.requestTemperatures(); // Send the command to get temperatures
  Serial.println("DONE");
  // After we got the temperatures, we can print them here.
  // We use the function ByIndex, and as an example get the temperature from the first sensor only.
  Serial.print("Temperature for the device 1 (index 0) is: ");
  Serial.println(sensors.getTempCByIndex(0));  
}
```

### Errors or Incorrect Output
It returns 85°C all the time.

Looking forward to help fix this issue soon.

LeoDJ, 2018-05-06:
**Update**: Today I debugged a bit further and I find it even weirder now.
The library definitely leaves the pin as output (last macro called is `DIRECT_MODE_OUTPUT()`) and also the correct bit in the `GPE` pin mode register is set to output (1) during the conversion.  
As far as I can see, the pin should be at 3.3V, but the oscilloscope definitely says the pin is dragged down too low (1.5V) during conversion. So it's only pulled up that far by the 4.7K pull-up resistor and not the GPIO pin for whatever reason...

I poked around here and there in the library but could not find the error. From what I can see, the software says that it should be working fine and my old binary says the hardware works fine too, but obviously it doesn't...

After five hours of unsuccessful debugging I gave up again. Maybe someone else can reproduce / has already experienced the issue and is not satisfied with the dirty workaround of using a lower pull-up resistor?

SwiCago, 2018-09-24:
Hi Leo, I am running latest IDE, latest esp library and latest onewire. With the demo sketches I get -127 or no more addresses, depending on sketch. Go to a different older firmware and it works again. Did you ever figure out the problem?

PaulStoffregen, 2018-09-25:
At the risk of repeating myself, I'm depending on the ESP community to submit pull requests for ESP8266 specific problems.  I do not directly support ESP and I can not get involved in these sorts of changes that happen on the ESP platform.

Dooque, 2018-10-08:
Hi, I'm experimenting the same problem.
@SwiCago could you please tell which particular version did work for you? Since it's ok for me to use an older version if that make it work, at least until the issue get fixed.
Thanks.

fguiet, 2019-01-26:
Hi, same thing here....could not DS18B20 sensor work with Arduino IDE 1.8.5 and lastest OneWire/Dallas library... I give up...

SwiCago, 2019-01-27:
@Dooque  @fguiet , I beieve we all got some cheap bootleg variation sensor. I ordered one from a sonoff vender directly and that one worked perfectly, even with a 3m extension cable.
Here is the link to the sensor that worked for me 
https://www.amazon.com/Freeshine-Sonoff-Waterproof-Temperature-Transmitter/dp/B078NRBNM8/ref=sr_1_1?ie=UTF8&qid=1548603536&sr=8-1&keywords=ds18b20+sonoff+th16

PaulStoffregen, 2019-01-27:
Any suggestions where to buy the not-so-good knockoff sensor?  (for testing with this library...)  All the ones I have are genuine, purchased from Digikey.

LeoDJ, 2019-01-27:
Hm, that might be a possiblity, but I for myself definitley have got genuine DS18B20s, because I ordered them directly from Maxim Integrated.

fguiet, 2019-01-27:
Some updates : got that sketch on my wemos which was working without a glitch for ages...decided to make some changes...and *boom* impossible to make it *great 😉* again...tried everything...tried also a bare ds18b20s chip. Decided to upload my sketch on a esp32 and guess what....its working like a charm with every ds18b20s I have. So, voilà, I had to replace my fellow wemos with this over powerful esp32...cry!!! But at least it works! Fred

Dooque, 2019-01-27:
Yes, all those alternatives sounds good but they are workarounds, I really would like to fix the problem because I already have the hardware. I didn't have the time to take a deeper look at this, I think my subconscious is intentionally avoiding it.

olirogers, 2019-07-31:
I have the same issue. Resurrected some old code that I knew worked a while ago and on Arduino IDE 1.8.9, OneWire 2.3.4, DallasTemperature 3.8.0 I get the same 85 degree issue. I had the circuit in powered (non-parasite) mode or parasite mode and got the same result.

Replacing the 4.7k resistor with a 1k resistor solves the issue but I am not sure how "safe" that is for the components (considering the datasheet specifically calls out 4.7k).

Not sure how I can help solve this one, let me know.

Harald2019F, 2019-08-17:
Hello, these days I also started with ESP8266 and temperature sensor DS18B20 84 dsensos on one wire) and I was very disappointed about the behaviour.  After some hours  all 4 sensors provided a value of -127; this happens very sporadically. I googled a lot  but I could not find any real solution. I am using the newest Arduino version and the newest Dalles one as well. Finally i am using a workaround: I check the value and if it is -127  and cut the power from the sensor via software (relais is controlling the power of the sensors and relais is controlled by Arduino pin).
Has anybody found a proper solution for this issue ?

Thank and regards
Harald

Robert-Arciszewski, 2019-09-25:
If you get -127, turn around DS18B20. It can be plugged incorrect side.

skandragon, 2019-10-09:
> @Dooque @fguiet , I beieve we all got some cheap bootleg variation sensor. I ordered one from a sonoff vender directly and that one worked perfectly, even with a 3m extension cable.
> Here is the link [...]

I am getting similar issues, and I got mine directly from the source, but a very long time ago.

mario0990, 2020-01-14:
I got the same problem, 85°C, but it turned out that in my code was missing one line - requesting temperature before reading temperature was done:
        sensor.requestTemperatures();
        delay(1000);
        objTemp = sensor.getTempCByIndex(0);

Mario

timwhite, 2020-04-11:
@mario0990 That fixed it for me too. It seems when using parasitic power we need to wait for that conversion to happen. The builtin delays in the library mustn't be long enough for the ESP8266 to provide the required power, so a longer delay gives that that chance. I also had to change down to a 1K resistor. I've seen library hacks pulling the pin HIGH while waiting, maybe that can be implemented in this library?

filipsPL, 2020-07-09:
Adding `delay` to the code, as @mario0990 proposed:

```
sensor.requestTemperatures();
delay(1000);
objTemp = sensor.getTempCByIndex(0);
```

as well as **downgrading** OneWire library from 2.3.5 to 2.3.0 solved this problem for me :tada:

sohnvonoff, 2020-08-24:
I am using several DS18b20s with parasite power. A 1k resistance is still too high. In my experience, 750-820 Ohm is optimal with one or more sensors.

plop3, 2020-08-25:
Same issue for me.
Downgrading OneWire library from 2.3.5 to 2.3.0 and it's work.

gclugston, 2020-10-25:
@plop3 - I have two computers, a desktop and laptop and was developing on the laptop which has Arduino 1.8.5.  Then wanted to develop on the desktop which runs 1.8.13 on ESP8266 Node MCU.  Suddenly was getting WDT Soft Reset on  sensors.begin();  on all my DS18B20 sketches.  You saved me a lot of time as I was checking sensors, connections, boards, power supplies, etc.  The problem was fixed by downgrading OneWire from 2.3.5 to 2.3.0.  Laptop uses IDE 1.8.5 with OneWire 2.3.5.   Desktop was using IDE 1.8.13 with OneWire 2.3.5.  Once I downgraded the desktop OneWire to 2.3.0 works great,

TriggerMeister, 2020-11-29:
If you are using PlatformIO you need to update your platformio.ini to point to the old OneWire 2.3 lib since it's not available in the GUI selection (only goes back to 2.3.2)

[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
lib_deps = 
	https://github.com/PaulStoffregen/OneWire/archive/2.3.zip
	milesburton/DallasTemperature@3.7.7
monitor_speed = 115200

lubino, 2021-04-29:
85°C on DS18B20 almost always means powering issues. Check you wiring, voltage, impedance, look for HW issues, not SW glitches.

skandragon, 2021-04-30:
> 85°C on DS18B20 almost always means powering issues. Check you wiring, voltage, impedance, look for HW issues, not SW glitches.

Sure, but then downgrading to a previous release should not cause this to be fixed, and people are reporting that it generally does.  It could also be explained by a change in timing in the software, where it is not providing sufficient time to allow the conversion to take place.

t-i-t-o, 2022-01-04:
Same problem hare - hooked 4 ds18b20 on 10m cable with external MOSFET strong pull up:
- with 4.7k neither sensor works i.e. 85C is read on all of them.
- with 1k the first sensor works only, others 85C.
- with delay(1000) three works, one not. Increasing delay to 2000 does not help.
- downgrading to 2.3.0 fixes everything, all four sensors are read without any delay() in code.

Kalougear, 2022-08-13:
I dont even know why it worked,but i had the same problem,I was reading my sensors with their hex addresses,i tryied everything you said and much more,then i just uploaded a sketch that reads them by index and suddenly all the sensors were warking fine,they i reflashed on by hex readings my main code and its just fixed....hmmmm

uzi18, 2022-08-13:
@Kalougear try OneWireNg