PaulStoffregen/OneWire #59: write_bit bug

fonsoreyes, 2018-05-02:
write_bit has a bug. Actually it writes 5V in the bus when it must allow it float.

void OneWire::write_bit(uint8_t v)
{
	IO_REG_TYPE mask IO_REG_MASK_ATTR = bitmask;
	volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;

	if (v & 1) {
		noInterrupts();
		DIRECT_WRITE_LOW(reg, mask);
		DIRECT_MODE_OUTPUT(reg, mask);	// drive output low
		delayMicroseconds(10);
		DIRECT_WRITE_HIGH(reg, mask);	// --> must be DIRECT_MODE_INPUT to allow it to float
		interrupts();
		delayMicroseconds(55);
	} else {
		noInterrupts();
		DIRECT_WRITE_LOW(reg, mask);
		DIRECT_MODE_OUTPUT(reg, mask);	// drive output low
		delayMicroseconds(65);
		DIRECT_WRITE_HIGH(reg, mask);	// --> must be DIRECT_MODE_INPUT to allow it to float
		interrupts();
		delayMicroseconds(5);
	}
}

This way the bus can be 3,3V or 5V.

Best regards

nomis, 2019-09-29:
`write_bit()` is documented as always leaving the bus powered; you need to call `depower()`

PaulStoffregen, 2019-10-03:
closing this as a duplicate of #60

closed 2019-10-03 by PaulStoffregen