PaulStoffregen/OneWire #63: error during compiling board Arduino/Genuino Uno

HarrieRouwhorst, 2018-11-09:
Please use this form only to report code defects or bugs.

For any question, even questions directly pertaining to this code, post your question on the forums related to the board you are using.

Arduino: forum.arduino.cc
Teensy: forum.pjrc.com
ESP8266: www.esp8266.com
ESP32: www.esp32.com
Adafruit Feather/Metro/Trinket: forums.adafruit.com
Particle Photon: community.particle.io

If you are experiencing trouble but not certain of the cause, or need help using this code, ask on the appropriate forum.  This is not the place to ask for support or help, even directly related to this code.  Only use this form you are certain you have discovered a defect in this code!

Please verify the problem occurs when using the very latest version, using the newest version of Arduino and any other related software.

----------------------------- Remove above -----------------------------

### Description

Describe your problem.

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

Board
Shields / modules used
Arduino IDE version
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

HarrieRouwhorst, 2018-11-09:
Sorry.  I made a mistake during typing the message.

**Description**

I'm trying compiling the example program // DS18S20 Temperature chip i/o.
Unfortunately. There are some messages and an error.

**Hardware & Software**

I have connected an Arduino Uno board to the computer.
My Arduino version is 1.8.3.
My OneWire version is installed August 13, 2018.
 
**The message of the compiler**
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Eigenaar\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10803 -build-path C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_117286 -warnings=all -build-cache C:\Users\Eigenaar\AppData\Local\Temp\arduino_cache_458378 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose D:\harm\electronica\arduino\Test_dallas_DS18S20\Test_dallas_DS18S20.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Eigenaar\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10803 -build-path C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_117286 -warnings=all -build-cache C:\Users\Eigenaar\AppData\Local\Temp\arduino_cache_458378 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose D:\harm\electronica\arduino\Test_dallas_DS18S20\Test_dallas_DS18S20.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_117286\sketch\Test_dallas_DS18S20.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\OneWire\src" "C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_117286\sketch\Test_dallas_DS18S20.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\OneWire\src" "C:\Program Files (x86)\Arduino\libraries\OneWire\src\OneWire.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\OneWire\src" "C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_117286\sketch\Test_dallas_DS18S20.ino.cpp" -o "C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_117286\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_117286\preproc\ctags_target_for_gcc_minus_e.cpp"
Sketch aan het compileren...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH [... cut ...]
     IO_REG_TYPE bitmask;

     ^

C:\Program Files (x86)\Arduino\libraries\OneWire\src/OneWire.h:61:14: error: 'IO_REG_TYPE' does not name a type

     volatile IO_REG_TYPE *baseReg;

              ^

Bibliotheek OneWire op versie 2.3.4 in map: C:\Program Files (x86)\Arduino\libraries\OneWire  wordt gebruikt
exit status 1
Fout bij het compileren van board Arduino/Genuino Uno

**Notes**
I think the issue is related to the declaration of IO_REG_TYPE. 
I don't understand this command.

Regards, Harrie Rouwhorst

PaulStoffregen, 2018-11-09:
> C:\Program Files (x86)\Arduino\libraries\OneWire\src/OneWire.h

Maybe you have a different OneWire library?  This one does not use a "src" folder.

Delete the bad OneWire library you have installed and install this one.  It works.

Or if you wish to use someone else's broken copy of OneWire, ask them for support.  Clearly you are **NOT** using this version of OneWire.  Don't open an issue here for someone else's code!

HarrieRouwhorst, 2018-11-10:
Paul, Thanks for your quick reply.

You were right I had installed the source code in the sub-folder src.
Today I have deleted all sub-folder and files from the OneWire library. Next I have downloaded your OneWire files by a ZIP file. 
And I have installed your library today again. And now in the libraries folder. See below.

**dir of OneWire Folder**

 De volumenaam van station C is C-schijf
 Het volumenummer is 28AD-5B56

 Map van C:\Program Files (x86)\Arduino\libraries\OneWire

10-11-2018  14:37    <DIR>          .
10-11-2018  14:37    <DIR>          ..
10-11-2018  14:36    <DIR>          docs
10-11-2018  14:36    <DIR>          examples
16-03-2018  00:07               819 keywords.txt
16-03-2018  00:07             1,423 library.json
16-03-2018  00:07               419 library.properties
16-03-2018  00:07            18,255 OneWire.cpp
16-03-2018  00:07             6,487 OneWire.h
10-11-2018  14:36    <DIR>          util
               5 bestand(en)           27,403 bytes
               5 map(pen)  122,239,090,688 bytes beschikbaar
**Next steps**

Next I have restarted the computer. Then I  compiled the example program again.
Unfortunately the same response.

**Error message**
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\OneWire\src" "C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_101739\sketch\Test_dallas_DS18S20.ino.cpp" -o "C:\Users\Eigenaar\AppData\Local\Temp\arduino_build_101739\sketch\Test_dallas_DS18S20.ino.cpp.o"
In file included from D:\harm\electronica\arduino\Test_dallas_DS18S20\Test_dallas_DS18S20.ino:1:0:

C:\Program Files (x86)\Arduino\libraries\OneWire\src/OneWire.h:60:5: error: 'IO_REG_TYPE' does not name a type

     IO_REG_TYPE bitmask;

     ^

C:\Program Files (x86)\Arduino\libraries\OneWire\src/OneWire.h:61:14: error: 'IO_REG_TYPE' does not name a type

     volatile IO_REG_TYPE *baseReg;

              ^

Bibliotheek OneWire op versie 2.3.4 in map: C:\Program Files (x86)\Arduino\libraries\OneWire  wordt gebruikt
exit status 1
Fout bij het compileren van board Arduino/Genuino Uno
Ongeldige bibliotheek gevonden in C:\Users\Eigenaar\Documents\Arduino\libraries\libraries: C:\Users\Eigenaar\Documents\Arduino\libraries\libraries
Ongeldige bibliotheek gevonden in C:\Users\Eigenaar\Documents\Arduino\libraries\TempertureSensorExample: C:\Users\Eigenaar\Documents\Arduino\libraries\TempertureSensorExample

**Still an error, but not the same**

Now the compiler replies : Invalid library found in (=Ongeldige bibliotheek gevonden in).
What did I wrong?

Regards, Harrie Rouwhorst

PaulStoffregen, 2018-11-10:
> What did I wrong?

The error *still* has "src" in the pathname.  Clearly you did not fully delete the bad copy.

HarrieRouwhorst, 2018-11-10:
Paul,
Thanks. I found the wrong library. 
Compiling was succesfull and the test program is running.

Regards, Harrie Rouwhorst

closed 2018-11-10 by PaulStoffregen