PaulStoffregen/OneWire #70: What pins number are supported?

nixmeer, 2019-05-16:
Hello,

which pin of the ESP8266 can handle a OneWire bus? I am using ESP12e and ESP12f

I could not find any documentation 

Best

lbussy, 2019-12-26:
This is not a OneWire specific question, but you can find more information [here](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/).

gjt211, 2022-02-03:
I am having a problem using GPIO16 on an ESP8266F module. It just will not work.
Using other GPIOs works fine but according to the link GPIO16 should still work as an input or output (which would be needed for DS18B20) and we don't need PWM or I2C, and not that I am a code expert but shouldn't need interrupts either.
Lots of searching on the web has not revealed any clear answer why this either will or won't work.

I am not using WiFi in my code, so if there is some esp core setting I can change or something else that can help would be super great.

gjt211, 2022-02-03:
I found a possible answer here in an open issue #27 and have modified my OneWire.h file so that the ARDUINO_ARCH_ESP8266 section is `#define PIN_TO_BITMASK(pin)             (1UL << pin)` instead of `#define PIN_TO_BITMASK(pin)             (1 << pin)` but it still doesn't work.