PaulStoffregen/OneWire #74: warning: unused variable 'reg'

brunohorta82, 2019-07-28:
.pio/libdeps/esp12e/OneWire_ID1/OneWire.cpp: In member function 'uint8_t OneWire::reset()':
.pio/libdeps/esp12e/OneWire_ID1/OneWire.cpp:167:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^
.pio/libdeps/esp12e/OneWire_ID1/OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':
.pio/libdeps/esp12e/OneWire_ID1/OneWire.cpp:201:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^
.pio/libdeps/esp12e/OneWire_ID1/OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':
.pio/libdeps/esp12e/OneWire_ID1/OneWire.cpp:229:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;

9a4gl, 2020-09-01:
Please merge nomis fix for this warning, same fix has to be applied to ESP32 as I actually had same warning on ESP32.

zfields, 2021-04-13:
Using v2.3.5 I ran into the same warning with the ESP32. It is easy to reproduce, add the Warnings: All flag to the Arduino IDE.

zeroject, 2023-02-09:
Please merge this branch as thoose viabels are not used. If someone has this problem open OneWire.cpp in your documents and delete "volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;" from the functions Readbit WriteBit and reset and your program should compile and work