PaulStoffregen/OneWire #76: ESP8266 and sensor DS18B20 provides -127 sporadically

Harald2019F, 2019-08-17:
Hello,
these days I started a project with ESP8266 and temperature sensor DS18B20 . I used 4 sensors on one wire) and I was very disappointed about the behaviour. After some hours all 4 sensors provided a value of -127; this happens very sporadically. Sometimes it happens after a few minutes, sometimes after a few hours. I checked the voltage of the data line of the sensors and noticed that whenever the error occurred it was about 1 V instead of 3.,3V. So, anythink must cause teh data line of the sensor putting to 1 V (low) constantly. Only a switching of the power of the sensor brings it back to a normal state. 
I am using  the newest Arduino version and the newest Dalles one as well. Finally I am using a workaround as follows: I check the values provided by the sensors and if it is -127 I am cutting the power from the sensor via software (relais is controlling the power of the sensors and relais is controlled by Arduino pin).

Has anybody found a proper solution for this issue ?

Thank and regards
Harald

skandragon, 2019-10-09:
I have a similar issue where I am getting a consistent -0.06 most of the time (all 0xff data except first byte) but sometimes I am getting -127.  I have 5 setups, all similar in  build, but this seems to occur mostly when I am using a DS18B20 that has a very short cable length (in this case, PCB traces.)  I do have a 4k7 resistor in use; I cannot easily change this at this point.

Humancell, 2019-10-09:
I've run into this periodically, and simply re-request the temperature and wait for the conversion to complete, and I usually get a good value.

Harald2019F, 2019-10-10:
In my case a re-request is not possible because the sensor is hanging up
from time to time. Only switching of that means disconnecting from the power
for a short time brings the sensor to working again. I implemented the
switching of process automatically via software when the program (Arduino)
recognize the hanging up of the sensor.

 

  _____  

-127 sporadically (#76)

 

<https://github.com/PaulStoffregen/OneWire/issues/76?email_source=notificati
ons&email_token=AM5UR3N5QN33MOMU32YQXN3QNZUW5A5CNFSM4IMQB642YY3PNVWWK3TUL52H
S4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEAZZWJY#issuecomment-540252967>
ZUW5ANCNFSM4IMQB64Q> .
<https://github.com/notifications/beacon/AM5UR3KFWW2HJXLXLA6GG3DQNZUW5A5CNFS
M4IMQB642YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEAZZWJY
.gif>

skandragon, 2019-10-17:
I solved this for me by giving someone else all my DS18B20-PAR devices, and just supplying power directly.

VeloSteve, 2024-09-12:
Just a comment.  We have sometimes had this problem with a system controlling water in experimental aquariums.

Annoyingly it would not show up during development on a clean desk or even measuring saltwater temps in a properly run lab.  As soon as we got "in the field" problems would increase.  It was mainly due to bad electrical conditions.  Be sure any devices or liquids touching your sensors are grounded at the same level as your controller and that the devices aren't badly built and "leaking" current.  I suspect we had some bad pumps or water heaters for a while.

In one remote spot we discovered that the ground pin of the wall outlets was putting out about 47 VAC relative to the wet steel-reinforced concrete floor we were standing on!  The water was grounded to the wall and was painful to touch.  A ground wire between the water and the floor reduced the pain and greatly improved the -127 problem.  Of course safe wiring in the building would be better, but we had about 3 days to collect data and no money for a repeat trip.

Sorry to be so chatty, but maybe this will help someone diagnose a problem.