PaulStoffregen/OneWire #81: STM32F103C8 compile error

AEmelyanov757, 2019-10-24:
Visual Studio Code version: 1.39.2
PlaformIO version: 4.0.3
OS: Windows_NT x64 10.0.18362

CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 5.6.0 > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)

Dependency Graph
|-- <DallasTemperature> 3.8.0
|   |-- <OneWire> 2.3.5
|   |   |-- <FreeRTOS701>
|-- <OneWire> 2.3.5
|   |-- <FreeRTOS701>
Compiling .pio\build\genericSTM32F103C8\lib7cd\OneWire_ID1\OneWire.cpp.o
In file included from C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:144:0:
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp: In member function 'void OneWire::begin(uint8_t)':
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:231:52: error: 'digitalPinToPinName' was not declared in this scope
 #define PIN_TO_BITMASK(pin)             ((uint32_t)digitalPinToPinName(pin))
                                                    ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:150:12: note: in expansion of macro 'PIN_TO_BITMASK'
  bitmask = PIN_TO_BITMASK(pin);
            ^~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:231:52: note: suggested alternative: 'digitalPinToBitMask'
 #define PIN_TO_BITMASK(pin)             ((uint32_t)digitalPinToPinName(pin))
                                                    ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:150:12: note: in expansion of macro 'PIN_TO_BITMASK'
  bitmask = PIN_TO_BITMASK(pin);
            ^~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp: In member function 'uint8_t OneWire::reset()':
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:55: error: 'PinName' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                       ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:55: note: suggested alternative: 'rename'
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                       ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:81: error: 'STM_MODE_INPUT' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                                 ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:81: note: suggested alternative: 'DIRECT_MODE_INPUT'
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                                 ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:97: error: 'GPIO_NOPULL' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                                                 ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:97: note: suggested alternative: 'GPIO_INPUT_PU'
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                                                 ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:68: error: 'STM_PIN_DATA' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
[... 588 lines cut ...]
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:68: error: 'STM_PIN_DATA' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                    ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:317:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(baseReg, bitmask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:68: note: suggested alternative: 'ADC_DR_DATA'
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                    ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:317:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(baseReg, bitmask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:41: error: 'pin_function' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                         ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:317:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(baseReg, bitmask);
  ^~~~~~~~~~~~~~~~~
C:\Users\aemel\.platformio\lib\OneWire_ID1\util/OneWire_direct_gpio.h:238:41: note: suggested alternative: '__section'
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                         ^
C:\Users\aemel\.platformio\lib\OneWire_ID1\OneWire.cpp:317:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(baseReg, bitmask);
  ^~~~~~~~~~~~~~~~~
*** [.pio\build\genericSTM32F103C8\lib7cd\OneWire_ID1\OneWire.cpp.o] Error 1

jhortwo, 2020-07-13:
Same problem for my code. Have you solved yet?

fpistm, 2021-01-07:
Hi,
OneWire library works fine with [STM32 core](https://github.com/stm32duino/Arduino_Core_STM32) using Arduino IDE.
I think you used the [maple](https://github.com/rogerclarkmelbourne/Arduino_STM32) core with PIO which does not include all those functions.
One thing strange is that `ARDUINO_ARCH_STM32` seems defined if you used the maple core.
Don't know if it can be linked to the fact PIO doesn't update the maple core as the `ARDUINO_ARCH_STM32` definition has been removed from it as it relies to the [STM32 core](https://github.com/stm32duino/Arduino_Core_STM32). For ref: https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/684
I think you should submit your issue to PIO: https://github.com/platformio/platform-ststm32/issues