PaulStoffregen/OneWire #89: [Question]: DS18S20 Sensor send Convertion ready?

hasenradball, 2020-02-06:
Hello Iam using the OneWire Library and I recognized that the delay after Convertion Request is set to fix values.
But the Ds18S20 send the ConvertionReady Information.
How can use or get this information?

Can I do it like this:
```
ds.reset();
ds.select(addr);
ds.write(0x44, 1);        // start conversion, with parasite power on at the end

/* Read Convertion ready info */
while(ds.read != 1) yield();
  
/* then not needed anymore  */
//delay(1000);     // maybe 750ms is enough, maybe not
// we might do a ds.depower() here, but the reset will take care of it.
  
present = ds.reset();
ds.select(addr);    
ds.write(0xBE);         // Read Scratchpad

Serial.print("  Data = ");
Serial.print(present, HEX);
```

PaulStoffregen, 2020-02-06:
This is the wrong place to ask questions.  The issue tracker is for reporting software defects.

Please use the forum for questions.

closed 2020-02-06 by PaulStoffregen

hasenradball, 2020-02-06:
Hi Do you have a link for the Forum?

Mit freundlichen Grüßen

Frank Häfele

Testato, 2020-02-06:
He mean Arduino forum