PaulStoffregen/OneWire #90: esp32 doesn't work

ramarro123, 2020-02-10:
esp32 seems to need some sort of workaround due to dualcore things.
i noticed a fork of your lib here
https://github.com/stickbreaker/OneWire

that "just work" with esp32

i know that the user haven't asked a pull request to merge his change, but do you plan to support esp32?

here the original issue if you need it for reference
https://github.com/espressif/arduino-esp32/issues/755

PaulStoffregen, 2020-02-10:
If "support" means merging a pull request with only edits the ESP32 defines in OneWire_direct_gpio.h and OneWire_direct_regtype.h, then yes, sure.  In fact, I've already merged such contributions in the past...

But if "support" means I will personally test ESP32 and get active involved in writing that code, then no.  I'm depending on the ESP community to submit pull requests.  The only reason things don't work perfectly today is the pull requests I've received and already merged weren't ideal.

garyd9, 2020-07-16:
Created PR #93 to merge stickbreaker's fork back to the origin master branch.  Tested on ESP32-WROVER and ESP32-WROOM.

9a4gl, 2020-09-08:
This is taking very long, can we get this in master and the as new version in platformio libraries ? There are many ESP32 devices arround waiting for this fix. I wasted a lot of time debugging until I have found I need to use alternative library, guess many people gave up as it does not work.

Gixy31, 2021-02-12:
I confirm the issue on ESP32/PlatformIO and that the modified library from stickbreaker solves the problem by masking interrupts during critical timing sections.

garyd9, 2021-02-12:
No commits to the project in over a year.  Hopefully @PaulStoffregen is still maintaining it.

Gixy31, 2021-02-13:
Hopefully...

altery, 2021-03-13:
This library currently does not work reliable with ESP32 and stickbreakers fork seems out of date. Hope the PR https://github.com/PaulStoffregen/OneWire/pull/93 can be merged! Would have saved me hours :)

klausj1, 2022-01-07:
For those who need a OneWire-Library working on ESP32 / ESP32-S2 (I only used it for ESP32-S2) Take [this one](https://github.com/pstolarz/OneWireNg), it is working fine, at least for me.

LenShustek, 2022-03-25:
I just wasted two days of debugging (with a logic analyzer, etc.) because I didn't know OneWire doesn't support ESP32. No error messages; it just doesn't work. 
Until it does, can at least a simple change be made so it generates a compile error for the ESP32 environment? This has been an unresolved issue for over two years. In the meantime I followed klausj1's suggestion and switched to OneWireNg.

uzi18, 2022-03-25:
This is best way

Jason2866, 2022-05-23:
Fix with #114

PaulStoffregen, 2022-05-23:
I've merged #114.  For everyone following this issue and still using ESP32, please download the latest OneWire and reply here to confirm (or deny) this fully solves the problem.

I am waiting for your feedback before publishing a new OneWire version which will make this fix available to everyone using the Arduino Library Manager.  Please take a moment to test this latest code and let me know if it should be published as a release to the wider Arduino community?