PaulStoffregen/OneWire #92: Arduino Nano BLE Sense does not work with OneWire

srv-config, 2020-03-27:
Please use this form only to report code defects or bugs.

For any question, even questions directly pertaining to this code, post your question on the forums related to the board you are using.

Arduino: forum.arduino.cc
Teensy: forum.pjrc.com
ESP8266: www.esp8266.com
ESP32: www.esp32.com
Adafruit Feather/Metro/Trinket: forums.adafruit.com
Particle Photon: community.particle.io

If you are experiencing trouble but not certain of the cause, or need help using this code, ask on the appropriate forum.  This is not the place to ask for support or help, even directly related to this code.  Only use this form you are certain you have discovered a defect in this code!

Please verify the problem occurs when using the very latest version, using the newest version of Arduino and any other related software.

----------------------------- Remove above -----------------------------

### Description

Describe your problem.

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

Board
Shields / modules used
Arduino IDE version
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

drdiesel, 2020-04-13:
I can verify OneWire is broken on the IoT.

fbartsch95, 2020-05-02:
Just to draw some attention to this: I am encountering the same problem as srv_config. I'm quite sure it's also related to the missing support of the nrf52840 based boards in OneWire, as described in this post: https://forum.arduino.cc/index.php?topic=673098.0

PaulStoffregen, 2020-05-11:
I'm depending on the open source community to contribute these I/O register defines.

Please feel free to volunteer.  Or at least keep following this issue.  Sooner or later someone familiar enough with the I/O registers in this chip will do it.  Then testing & confirmation by other people using this board will be needed.

uzi18, 2020-05-11:
@PaulStoffregen isn't it already covered by  #33 ?

RedaBousbah, 2021-04-09:
The library kinda works, but after a few reads makes the whole Arduino crash.

matevy, 2023-12-12:
I tried library 2.3.7 on Nano BLE and it did not work at all. After creating a support for the board (actually NRF52840) it works good.

Shall I contribute it / is this of any interest at all?