PaulStoffregen/OneWire #100: ESP32 compile error with arduino-esp32 v2.0.0-rc1 (error: 'rtc_gpio_desc' was not declared in this scope)

avillacis, 2021-08-02:
### Description

The OneWire library (latest version 2.3.5) fails to compile with arduino-esp32 v2.0.0-rc1 due to undeclared symbol 'rtc_gpio_desc'. This symbol apparently does not exist anymore under arduino-esp32 v2.0.0-rc1, and has been renamed to rtc_io_desc.

### Steps To Reproduce Problem

1. Install Arduino IDE
2. Install ESP32 support for Arduino, 2.0.0-rc1 (required for ESP32-S2 and ESP32-C3 support)
3. Install this library and some library requiring it
4. Compile any sketch that pulls OneWire (no need to flash anything, as this is a compile issue)

### Hardware & Software

Board: ESP32 (YUBOX)
Arduino IDE version: 1.8.15
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager): esp32 by Espressif systems v2.0.0-rc1
Operating system & version: Fedora 34 x86_64
Any other software or hardware?

### Arduino Sketch

```cpp
#include <Arduino.h>
// Pulling header file is enough to trigger the compile error
#include <OneWire.h>

void setup() {
// No code required to trigger error
}

void loop() {
// No code required to trigger error
}
```

### Errors or Incorrect Output
```
In file included from /home/alex/Arduino/libraries/OneWire/OneWire.cpp:144:
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:161:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:161:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:189:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
/home/alex/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:189:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
```

lpotter, 2021-08-03:
I am seeing this as well.

LordIkol, 2021-08-10:
first of all big thanks for reporting this. 
I have the same issue with NodeMCU ESP-32S Board.

Find rtc_gpio_desc replace with rtc_io_desc in the OneWire_direct_gpio.h file solved the issue

Ralms, 2021-08-21:
> first of all big thanks for reporting this.
> I have the same issue with NodeMCU ESP-32S Board.
> 
> Find rtc_gpio_desc replace with rtc_io_desc in the OneWire_direct_gpio.h file solved the issue

This worked for me on an ExpressIf ESP32-S2-WROVER.

avillacis, 2021-09-01:
> This worked for me on an ExpressIf ESP32-S2-WROVER.
> 
> Edit:
> Ok, it did not work.
> Although this allowed the code to compile, it broke OneWire functionality such as reading DS18B20 sensors.
> The problem is only with V2.0.0-rc1, as going back to 1.0.6, I was able to read the temperatures from the sensors straight away without any changes.

@Ralms Are you sure this was a ESP32-S2 board? The 1.0.6 version of Arduino-ESP32 does not ship support for the S2. Maybe you meant an ESP32-WROVER (no -S2).

The stable 2.0.0 version of Arduino-ESP32 just shipped yesterday with not many significant changes over -rc2. I presume the bug is still open for 2.0.0.

> first of all big thanks for reporting this.
> I have the same issue with NodeMCU ESP-32S Board.
> 
> Find rtc_gpio_desc replace with rtc_io_desc in the OneWire_direct_gpio.h file solved the issue

@LordIkol Could you please confirm whether merely fixing the symbol actually allowed the board to work correctly, not just compile, on 2.0.0-rcX or 2.0.0 stable? Your report and the one from @Ralms are in conflict.

altmannmarcelo, 2021-09-01:
I second @Ralms here. Changing  `rtc_gpio_desc` to `rtc_io_desc` makes it compile, but I always read temperature as:
```
16:13:46.584 -> -127.00ºC
16:13:46.584 -> -196.60ºF
``` 
Board ESP32-S2 
Arduino, 2.0.0 (GA released yesterday)

avillacis, 2021-09-01:
@altmannmarcelo The question of whether the OneWire pin programming is valid for ESP32-S2 might be a separate issue from whether it can still work correctly for ESP32 (the previous hardware platform) with just a symbol swap. Do you have an ESP32 (not an -S2) to test your sketch with?

altmannmarcelo, 2021-09-01:
@avillacis no unfortunately only S2 and ESP8266

Ralms, 2021-09-01:
> @Ralms Are you sure this was a ESP32-S2 board? The 1.0.6 version of Arduino-ESP32 does not ship support for the S2. Maybe you meant an ESP32-WROVER (no -S2).
>
> The stable 2.0.0 version of [...]

100%, its an original ExpressIf ESP32-S2-WROVER dev board.
My tests with the S2 were with Arduino IDE 1.8.15 and V2.0.0-rc1.

After this, I decided to try and use another board, ESP32-WROOM-32 from Az-Delivery, which also didn't work with V2.0.0-rc1, it had the same symptoms, would compile but unable to read sensors. 
Leaving me to believe to be some bigger change on V2 that is breaking some of these libs.

Using ESP32-WROOM-32 + ESP32 1.0.6 boards, everything worked normally again using the exact same code (without the rtc_gpio_desc replacement that is, I deleted and reinstalled the OneWire).

pstolarz, 2021-09-02:
[This lib](https://github.com/pstolarz/OneWireNg) works on all types of ESP32 + v2 SDK. Compatible with OneWire lib via `OneWire.h` header, therefore shall work out of the box.

Ralms, 2021-09-02:
> [This lib](https://github.com/pstolarz/OneWireNg) works on all types of ESP32 + v2 SDK. Compatible with OneWire lib via `OneWire.h` header, therefore shall work out of the box.

I will give it a try.
I can see that you are one of the maintainers, so if you don't mind me asking, why fork OneWire instead of expanding the existing lib?

pstolarz, 2021-09-02:
@Ralms  It's not a fork. The lib was written from scratch with its own API, but with compatibility interface with OneWire.

VladimirVecera, 2021-09-07:
Today I installed a new version of ESP32, I think 2.0.0 came out and OneWire worked without a problem before that and now I have an error. Is there a way to fix this bug so I don't have to rework the entire code? I use DS18B20 on multiple pins, so I use OneWire.h together with another library <DallasTemperature.h> 

**I have ESP32-WROOM-32U**

`In file included from c:\Users\jsem\Documents\Arduino\libraries\OneWire\OneWire.cpp:144:
c:\Users\jsem\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
c:\Users\jsem\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:161:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
c:\Users\jsem\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:161:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
c:\Users\jsem\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
c:\Users\jsem\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:189:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
c:\Users\jsem\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:189:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
Compilation error: Error: 2 UNKNOWN: exit status 1
`

ClemensGruber, 2021-09-07:
Same here code has worked with the 1.x ESP32 Arduino core but does not work with 2.x, I'm using the latest 2.x version 2.0.0 what is mared as "stable" in Arduino's board manager. So it would be nice in case this could be fixed.

PaulStoffregen, 2021-09-08:
> So it would be nice if this could be fixed.

Please send a pull request.  Hopefully you can find a preprocessor macro to check if the new or old ESP32 core is used?  Then just ifdef the 2 cases in the header files depending on which core library version.

Just to be clear, I'm depending on the open source community to submit pull requests for this sort of ESP32 support.

ClemensGruber, 2021-09-08:
I think we do not have a solution for core 2.x because the initial postulated 

> Find rtc_gpio_desc replace with rtc_io_desc in the OneWire_direct_gpio.h file solved the issue

does not work, it compiles but https://github.com/PaulStoffregen/OneWire/issues/100#issuecomment-910609906 the output is wrong, so a preprocessor macro to check ESP32 core version (1.x vs. 2.x) is not sufficient as I see. Or did I miss the solution?

lcsjunior, 2021-09-19:
I have the same problem. I recently bought a LOLIN S2 mini ESP32-S2.
I did that and it didn't either: Changing rtc_gpio_desc to rtc_io_desc makes it compile.
Any solution?
I think I will return the boards to the manufacturer. I'm wasting time.

ClemensGruber, 2021-09-19:
There is a hint from Ivan at https://github.com/espressif/arduino-esp32/issues/5641#issuecomment-915070581 

> I think can be fixed by enabling CONFIG_RTCIO_SUPPORT_RTC_GPIO_DESC=y [in sdkconfig](https://github.com/espressif/arduino-esp32/blob/92ce408f4c71d235f0961a265e8d202f9fd570d5/tools/sdk/esp32/sdkconfig#L455).

But I'm not familiar what to do exactly and how sdkconfig works. Afaik it is a compiling config, and you have to do it before ... perhaps someone can say what to do with sdkconfig and how to precede.

codepam, 2021-09-24:
Could you have a look at this pull request #102 and confirm that is ok?
It compiles without any problems and seems to be working on hardware.

nuclearcat, 2021-09-28:
Worked for me (no more compile errors, code works). Thanks for fix!
P.S. Github doesnt allow to download raw patches for PR, seems Microsoft demand to download their proprietary tools. Quite irritating :(

Ralms, 2021-09-28:
> Worked for me (no more compile errors, code works). Thanks for fix!
> P.S. Github doesnt allow to download raw patches for PR, seems Microsoft demand to download their proprietary tools. Quite irritating :(

I don't know where you got that idea, you can download the raw file from anywhere and it has nothing to do with Microsoft.
Its a bit out of topic, but just going to leave the instructions here so others don't get confused. 

In this case, all you really need is to go to the pull request, "Files Changed" tab. 
Then on each file changed (in this case is just one), on the top right corner you do "View File"
[image]

Lastly, see the Raw file and save normally, applying the correct file extension.
[image]

If you want to download the all the files including the changed ones at once, you have to download the entire branch. 
For this unfortunately, you need to use Git locally and clone the repository. 

You go to the branch where the files are coming from on the Pull Request, in this case: 
[image]

Grab the Repository URL todo "Git clone" locally, either specifying the branch on the clone or switching after.
The fastest way would be to do "git clone -b rtc_gpio_desc-not_declared https://github.com/codepam/OneWire.git"
[image]

GitHub has been like this since the beginning, there is nothing new.

ClemensGruber, 2021-09-28:
> Could you have a look at this pull request #102 and confirm that is ok?

Thanks @codepam for this pull request. 

Just as a recap Paul's current OneWire lib version 2.3.5 is working fine with Arduino core 1.0.6, an update to Arduino core 2.0.0 leads to the compiling errors above. 

So I updated to Arduino core 2.0.0, and changed the file 

OneWire/util/OneWire_direct_gpio.h 

according to https://github.com/PaulStoffregen/OneWire/pull/102 / https://github.com/PaulStoffregen/OneWire/blob/a9094d3e8609dd56e61126a501d33b23f6301509/util/OneWire_direct_gpio.h

Result:
- no compiling error 
- functional working, got reliable and valide temp data (in comparison to the other trials here in the thread)

So I can confirm that it is working but I did no code review.

codepam, 2021-09-28:
> Paul's current OneWire lib version 2.3.5 is working fine with Arduino core 1.0.6, an update to Arduino core 2.0.0 leads to the compiling errors above

@ClemensGruber that is why the patch is compatible with both versions, so as to be included without any problems as soon as possible. Currently, the library is unusable with arduino-esp32 v2.0.0.

Thanks for testing.

nuclearcat, 2021-09-29:
> 
> GitHub has been like this since the beginning, there is nothing new.
I'm quite sure before i was able to download patch and there was a button for this.
It still works if i craft URL manually, in this case:
https://github.com/codepam/OneWire/commit/a9094d3e8609dd56e61126a501d33b23f6301509.patch
or https://patch-diff.githubusercontent.com/raw/PaulStoffregen/OneWire/pull/102.patch

PaulStoffregen, 2021-09-30:
To everyone who commented or contributed on this issue, please give the latest code a try.  Does it fully solve the problem for you?  Does it still work with the older ESP software?

closed 2021-09-30 by PaulStoffregen

ClemensGruber, 2021-09-30:
> Does it still work with the older ESP software?

For testing I downgraded to Arduino ESP core 1.0.6 and code is still compiling and DS18B20 working as expected.

PaulStoffregen, 2021-10-04:
I'm waiting on more ESP32 user feedback before publishing a new OneWire version.  So far only 3 people have reported results with #102 applied.

Michal-Szczepaniak said "+1 works perfectly on my esp32"
damianjwilliams said "Not working on my ESP32-S2-WROVER"
ClemensGruber said "I downgraded to Arduino ESP core 1.0.6 and code is still compiling and DS18B20 working as expected".

Now is the time to contribute by just installing the latest and running on your ESP32 hardware, then post clear feedback.  Is this ready to publish as a new version to Arduino's library manager?

ClemensGruber, 2021-10-04:
Just to clarify: Along to https://github.com/PaulStoffregen/OneWire/issues/100#issuecomment-929140837 I tested against Arduino core 2.0.0 and the new code is working, in https://github.com/PaulStoffregen/OneWire/issues/100#issuecomment-931427375 I tested downward compatibility and also this is ok in my test.

PaulStoffregen, 2021-11-18:
Done  https://github.com/PaulStoffregen/OneWire/releases/tag/v2.3.6

Hopefully the problems reported above with certain boards aren't common.  Again, I don't test ESP hardware and I'm depending on the ESP community to sent pull requests for this sort of thing.

PR #102 (2021-09-30): Fix 'rtc_gpio_desc was not declared' compile error with arduino-esp32 v2.0.0
When using arduino-esp32 v2.0.0, the call to rtc_gpio_desc is replaced by rtc_io_desc and the io number is assigned by rtc_io_number_get.

Fixes #100
Fixes #103