PaulStoffregen/OneWire #103: Getting Guru Meditation Error when using DallasTemperature && OneWire with Wifi.h on esp32

Soumi422, 2021-09-26:
I'm getting this error, when trying to use Wifi + DallasTemperature && OneWire on my esp32. Temperature sensor is Gravity DS18B20.

 When I don't use Wifi it works fine.

This is an error code

```
09:41:17.049 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
09:41:17.082 -> 
09:41:17.082 -> Core  1 register dump:
09:41:17.082 -> PC      : 0x400d3b40  PS      : 0x00060433  A0      : 0x800d4488  A1      : 0x3ffb2710  
09:41:17.082 -> A2      : 0x3f40af28  A3      : 0x0000001c  A4      : 0x0a4c4c55  A5      : 0x3ff49000  
09:41:17.082 -> A6      : 0x00000020  A7      : 0x00000000  A8      : 0x800d3b15  A9      : 0x3ffb26d0  
09:41:17.082 -> A10     : 0x00060420  A11     : 0x3ffc2cac  A12     : 0x3ffc2ca4  A13     : 0xb33fffff  
09:41:17.082 -> A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
09:41:17.115 -> EXCVADDR: 0x0a4c4c55  LBEG    : 0x400df328  LEND    : 0x400df332  LCOUNT  : 0x00000000  
09:41:17.115 -> 
09:41:17.115 -> 
09:41:17.115 -> Backtrace:0x400d3b3d:0x3ffb27100x400d4485:0x3ffb2750 0x400d476c:0x3ffb2780 0x400d2782:0x3ffb27b0 0x400d53b7:0x3ffb2820 

```

Here I have it decoded

```
Decoding 5 results
0x400d3b3d: OneWire::reset() at /home/suomi/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h line 165
:  (inlined by) OneWire::reset() at /home/suomi/Arduino/libraries/OneWire/OneWire.cpp line 172
0x400d4485: OneWire::search(unsigned char*, bool) at /home/suomi/Arduino/libraries/OneWire/OneWire.cpp line 388
0x400d476c: DallasTemperature::begin() at /home/suomi/Arduino/libraries/DallasTemperature/DallasTemperature.cpp line 113
0x400d2782: setup() at /home/suomi/Arduino/sketch_sep18a/sketch_sep18a.ino line 53
0x400d53b7: loopTask(void*) at /home/suomi/.arduino15/packages/esp32/hardware/esp32/2.0.0/cores/esp32/main.cpp line 38
```

also with backtracked lines in the files mentioned above

```
1. ESP_REG(rtc_reg) = ESP_REG(rtc_reg) & ~(rtc_io_desc[pin].mux);

2. DIRECT_MODE_INPUT(reg, mask);

3.if (!reset()) {

4.while (_wire->search(deviceAddress)) {

5.sensors.begin();

6.void loopTask(void *pvParameters)
{
    setup();

``` 

And my executed code

```
#include <OneWire.h>
#include <DallasTemperature.h>
#include <WiFi.h>

// Wifi Settings
#define WIFI_NETWORK "name"
#define WIFI_PASSWORD "pass"
#define WIFI_TIMOUT_MS 20000

const int oneWireBus = 32;
OneWire oneWire(oneWireBus);
DallasTemperature sensors(&oneWire);
int status;

int connectToWiFi() {
    Serial.println(" ");
    Serial.print("Connencting to WiFi");
    Serial.print(" ");
    WiFi.mode(WIFI_STA);
    WiFi.begin(WIFI_NETWORK, WIFI_PASSWORD);

    unsigned long startAttemptTime = millis();

    while(WiFi.status() != WL_CONNECTED && millis() - startAttemptTime < WIFI_TIMOUT_MS) {
        Serial.print(".");
        delay(700);
    }

    if (WiFi.status() != WL_CONNECTED) {
        Serial.println("Failed connect to WiFi!");
        return 0;

    }

    else {
        Serial.print("Connected to WiFi");
        Serial.println(WIFI_NETWORK);
        Serial.println(WiFi.localIP());
        return 1;
    }
}

void setup() {
    status = connectToWiFi();
    Serial.begin(115200);
    sensors.begin();

void loop() {
        
}
```

I'm not very good in c++, could someone please help me? `LoadProhibited`: It looks like it trying to access a memory that it shouldn't?

codepam, 2021-09-26:
It seems you are using arduino-esp32 v2.0.0 and you have replaced `rtc_gpio_desc` with `rtc_io_desc` in util/OneWire_direct_gpio.h.
Could you test the changes from https://github.com/PaulStoffregen/OneWire/pull/102/files instead?
Thanks.

Soumi422, 2021-09-26:
> It seems you are using arduino-esp32 v2.0.0 and you have replaced `rtc_gpio_desc` with `rtc_io_desc` in util/OneWire_direct_gpio.h.
> Could you test the changes from https://github.com/PaulStoffregen/OneWire/pull/102/files instead?
> Thanks.

@codepam Thanks! This patch worked with the newest version without error!

codepam, 2021-09-26:
@Soumi422 thank you for confirming.

Duplicate of #100 .

codepam, 2021-09-26:
Duplicate of #100

closed 2021-09-30 by PaulStoffregen

PaulStoffregen, 2021-10-04:
I'm waiting on more ESP32 user feedback before publishing a new OneWire version.  So far only 3 people have reported results with #102 applied.

Michal-Szczepaniak said "+1 works perfectly on my esp32"
damianjwilliams said "Not working on my ESP32-S2-WROVER"
ClemensGruber said "I downgraded to Arduino ESP core 1.0.6 and code is still compiling and DS18B20 working as expected".

Now is the time to contribute by just installing the latest and running on your ESP32 hardware, then post clear feedback.  Is this ready to publish as a new version to Arduino's library manager?

PR #102 (2021-09-30): Fix 'rtc_gpio_desc was not declared' compile error with arduino-esp32 v2.0.0
When using arduino-esp32 v2.0.0, the call to rtc_gpio_desc is replaced by rtc_io_desc and the io number is assigned by rtc_io_number_get.

Fixes #100
Fixes #103