PaulStoffregen/OneWire #105: Arduino Nano RP 2040 error

Michal-Visuality, 2021-12-22:
### Description

Describe your problem.

When uploading sample script to Arduino Nano RP 2040 the arduino crashes:(

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

Board: Arduino Nano RP 2040
Shields / modules used: ds18b20
Arduino IDE version 1.8.18 (same problem with webIDE)
Version info & package name (from Tools > Boards > Board Manager) : VID: 2341 PID: 005e
Operating system & version: MACOS 12.1
Any other software or hardware? Nope.

### Arduino Sketch

```cpp
#include <OneWire.h>

// OneWire DS18S20, DS18B20, DS1822 Temperature Example
//
// http://www.pjrc.com/teensy/td_libs_OneWire.html
//
// The DallasTemperature library can do all this work for you!
// https://github.com/milesburton/Arduino-Temperature-Control-Library

OneWire  ds(10);  // on pin 10 (a 4.7K resistor is necessary)

void setup(void) {
  Serial.begin(9600);
}

void loop(void) {
  byte i;
  byte present = 0;
  byte type_s;
  byte data[12];
  byte addr[8];
  float celsius, fahrenheit;
  
  if ( !ds.search(addr)) {
    Serial.println("No more addresses.");
    Serial.println();
    ds.reset_search();
    delay(250);
    return;
  }
  
  Serial.print("ROM =");
  for( i = 0; i < 8; i++) {
    Serial.write(' ');
    Serial.print(addr[i], HEX);
  }

  if (OneWire::crc8(addr, 7) != addr[7]) {
      Serial.println("CRC is not valid!");
      return;
  }
  Serial.println();
 
  // the first ROM byte indicates which chip
  switch (addr[0]) {
    case 0x10:
      Serial.println("  Chip = DS18S20");  // or old DS1820
      type_s = 1;
      break;
    case 0x28:
      Serial.println("  Chip = DS18B20");
      type_s = 0;
      break;
    case 0x22:
      Serial.println("  Chip = DS1822");
      type_s = 0;
      break;
    default:
      Serial.println("Device is not a DS18x20 family device.");
      return;
  } 

  ds.reset();
  ds.select(addr);
  ds.write(0x44, 1);        // start conversion, with parasite power on at the end
  
  delay(1000);     // maybe 750ms is enough, maybe not
  // we might do a ds.depower() here, but the reset will take care of it.
  
  present = ds.reset();
  ds.select(addr);    
  ds.write(0xBE);         // Read Scratchpad

  Serial.print("  Data = ");
  Serial.print(present, HEX);
  Serial.print(" ");
  for ( i = 0; i < 9; i++) {           // we need 9 bytes
    data[i] = ds.read();
    Serial.print(data[i], HEX);
    Serial.print(" ");
  }
  Serial.print(" CRC=");
  Serial.print(OneWire::crc8(data, 8), HEX);
  Serial.println();

  // Convert the data to actual temperature
  // because the result is a 16 bit signed integer, it should
  // be stored to an "int16_t" type, which is always 16 bits
  // even when compiled on a 32 bit processor.
  int16_t raw = (data[1] << 8) | data[0];
  if (type_s) {
    raw = raw << 3; // 9 bit resolution default
    if (data[7] == 0x10) {
      // "count remain" gives full 12 bit resolution
      raw = (raw & 0xFFF0) + 12 - data[6];
    }
  } else {
    byte cfg = (data[4] & 0x60);
    // at lower res, the low bits are undefined, so let's zero them
    if (cfg == 0x00) raw = raw & ~7;  // 9 bit resolution, 93.75 ms
    else if (cfg == 0x20) raw = raw & ~3; // 10 bit res, 187.5 ms
    else if (cfg == 0x40) raw = raw & ~1; // 11 bit res, 375 ms
    //// default is 12 bit resolution, 750 ms conversion time
  }
  celsius = (float)raw / 16.0;
  fahrenheit = celsius * 1.8 + 32.0;
  Serial.print("  Temperature = ");
  Serial.print(celsius);
  Serial.print(" Celsius, ");
  Serial.print(fahrenheit);
  Serial.println(" Fahrenheit");
}
```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears

In file included from /Users/michalpiorkowski/Documents/Arduino/libraries/OneWire-master/OneWire.cpp:144:0:
/Users/michalpiorkowski/Documents/Arduino/libraries/OneWire-master/util/OneWire_direct_gpio.h:459:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
 #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
  ^~~~~~~
/Users/michalpiorkowski/Documents/Arduino/libraries/OneWire-master/OneWire.cpp: In member function 'uint8_t OneWire::reset()':
/Users/michalpiorkowski/Documents/Arduino/libraries/OneWire-master/OneWire.cpp:167:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~
/Users/michalpiorkowski/Documents/Arduino/libraries/OneWire-master/OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':
/Users/michalpiorkowski/Documents/Arduino/libraries/OneWire-master/OneWire.cpp:201:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~
/Users/michalpiorkowski/Documents/Arduino/libraries/OneWire-master/OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':
/Users/michalpiorkowski/Documents/Arduino/libraries/OneWire-master/OneWire.cpp:229:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~

pstolarz, 2021-12-31:
As indicated by `"OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."`, this platform is not supported by OneWire lib.

BTW: Since RP 2040 chip is becoming quite popular I've added support for it in my [OneWireNg](https://github.com/pstolarz/OneWireNg) library. I don't have any RP 2040 based platform, so I'm not able to check the implementation correctness. Fell free to check it out on your board. The library is compliant with OneWire via `OneWire.h` C++ header therefore shall work out of the box after replacing OneWire with OneWireNg.

gbr1, 2023-06-06:
Hi @Michal-Visuality, check this [PR127](https://github.com/PaulStoffregen/OneWire/pull/127)