PaulStoffregen/OneWire #108: Arduino-ESP32 developers changed API (GPIO now using ESP-IDF API on all chips.)

dukess, 2022-02-27:
This will cause compile error with fresh versions of the SDK.

```
In file included from /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
from src/src/OneWire/OneWire.cpp:147:
src/src/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
src/src/OneWire/util/OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope
ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
^~~~~~~~~~~~~
/home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
#define ESP_REG(addr) *((volatile uint32_t *)(addr))
^~~~
src/src/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
src/src/OneWire/util/OneWire_direct_gpio.h:231:38: error: 'esp32_gpioMux' was not declared in this scope
ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
^~~~~~~~~~~~~
/home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
#define ESP_REG(addr) *((volatile uint32_t *)(addr))
^~~~
*** [.pio/build/esp32-evb/src/src/OneWire/OneWire.cpp.o] Error 1
```

Diff: 
https://github.com/espressif/arduino-esp32/commit/c7cc5c90eb397b040568d1eb2d04b6bd3601b136

uzi18, 2022-02-27:
Or switch to OneWireNg

dukess, 2022-02-28:
@uzi18 It can be good idea, but project (https://github.com/fredlcore/BSB-LAN) right now work on some platforms: Mega, Due, ESP32. OneWireNg not tested yet with Due (SAM).

uzi18, 2022-02-28:
https://github.com/pstolarz/OneWireNg/blob/master/src/platform/OneWireNg_ArduinoSAM.h

dukess, 2022-03-01:
Ok, thank you!
We should to discuss this idea with main developer.

Jason2866, 2022-05-23:
PR #114

PaulStoffregen, 2022-05-23:
I've merged #114.  For everyone following this issue and still using ESP32, please download the latest OneWire and reply here to confirm (or deny) this fully solves the problem.

I am waiting for your feedback before publishing a new OneWire version which will make this fix available to everyone using the Arduino Library Manager.  Please take a moment to test this latest code and let me know if it should be published as a release to the wider Arduino community?

brownrb, 2022-06-13:
Hi Paul
My apologies if I have not followed any predefined formats for this message. Age and memory loss is working against me. 

I downloaded and used OneWire 114 to replace existing library as I have recently started to have compile issues with my projects. These projects have been around for a few years now so are well established. 

The compile issues concern esp32 and OneWire (as stated, fine for a few years but now broken with recent core esp32 changes). I am using 
Arduino IDE 1.8.19
esp32 core 2.0.3
Target Esp32 Dev Module

I am still unable to compile clean using esp32 core 2.0.3 (ESP32 Dev) and OneWire 114 as listed below, same issue

I have found threads related to recent issues about ESP32 and OneWire. A number of my projects use OneWire and ESP32 but no longer compile now, I only know that the code will no longer compile. 

(I can confirm that replacing OneWire with the OneWire-Stickbreaker compiles clean and runs.)

The code did not have an issue before the latest round of updates for the ESP32 Core. 
In particular, I use a shared Mutex around the use of the temperature update code, for example
```
      portENTER_CRITICAL(&tempMux);
      t_mux = update_temp_flag;
      portEXIT_CRITICAL(&tempMux);
      if ( t_mux == 1 )
      {
        portENTER_CRITICAL(&tempMux);
        update_temp_flag = 0;
        portEXIT_CRITICAL(&tempMux);
        runtime_println("TMP-2");
        temp = tempprobe->update();
      }
```
but I include that purely for reference as this code is unchanged from earlier versions of my code.  
Am happy to test code/fixes etc at any time. 

Error Messages from Arduino Compiler
```
Compiling library "OneWire"
"C:\\Users\\rob\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.1-1-gb8050b365e\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\rob\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32/include/config" "-

IC:\\Users\\rob\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3\\libraries\\Wire\\src" "-ID:\\Documents\\Arduino\\libraries\\OneWire" "-ID:\\Documents\\Arduino\\libraries\\myDallasTemperature" "-IC:\\Users\\rob\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3\\libraries\\WebServer\\src" "-ID:\\Documents\\Arduino\\libraries\\myOLED" "D:\\Documents\\Arduino\\libraries\\OneWire\\OneWire.cpp" -o "D:\\TEMPUSER\\arduino_build_624369\\libraries\\OneWire\\OneWire.cpp.o"
In file included from C:\Users\rob\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/Arduino.h:36,
                 from D:\Documents\Arduino\libraries\OneWire\OneWire.cpp:142:
D:\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
D:\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope
         ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
                                      ^~~~~~~~~~~~~
C:\Users\rob\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~
D:\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
D:\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:232:38: error: 'esp32_gpioMux' was not declared in this scope
         ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
                                      ^~~~~~~~~~~~~
C:\Users\rob\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~

```

regards
Robert

PaulStoffregen, 2022-06-13:
First quit Arduino.  Try deleting all copies of OneWire.  Then restart Arduino use Arduino's library manager to install version 2.3.7.  Does that make any difference?

PaulStoffregen, 2022-06-13:
Do all of the examples provided with OneWire compile without error?  Is the error only when compiling with your program?

brownrb, 2022-06-13:
Hi Paul
Thank you for the speedy response.
I can confirm that the solution you outlined has resolved this issue for me. Thank you.
(ie, delete library, re-install from Library Manager)

Regards
Robert

brownrb, 2022-08-05:
Back again.
I upgraded from esp32 core 2.0.3 to 2.0.4 yesterday and the OneWire issues have come back. Different error this time. 

I followed again, the previous advice above, deleted all OnWire library, etc, updated to Library 2.3.7 using Library Manager in Arduino IDE, but the errors still remain.
I confirm that the OneWire example DS18x20_Temperature also did not compile and generates the same compiler errors, which is

`Compiling library "OneWire"

D:\Documents\Arduino\libraries\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::reset()':
D:\Documents\Arduino\libraries\OneWire\OneWire.cpp:167:24: error: unused variable 'reg' [-Werror=unused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~
D:\Documents\Arduino\libraries\OneWire\OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':
D:\Documents\Arduino\libraries\OneWire\OneWire.cpp:201:24: error: unused variable 'reg' [-Werror=unused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~
D:\Documents\Arduino\libraries\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':
D:\Documents\Arduino\libraries\OneWire\OneWire.cpp:229:24: error: unused variable 'reg' [-Werror=unused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~
cc1plus.exe: some warnings being treated as errors
Using library OneWire at version 2.3.7 in folder: D:\Documents\Arduino\libraries\OneWire 
`

If you have any suggestions I would appreciate that, and again, am happy to do any testing as I mostly now use esp32. On a side note, I changed the target CPU to Arduino Nano and get compiler warnings when compiling the DS18x example.

regards
Robert

PaulStoffregen, 2022-08-05:
Don't use -Werror

brownrb, 2022-08-05:
Hi Paul
Thanks for the reply.
I take it that -Werror comes from the compiler, and the suggestion is to turn off compiler error settings (change the setting I have it set to) in the Arduino IDE.
And it compiles again (but somehow I really do like seeing all those messages scrolling through, it becomes addictive after a while)
regards
Robert

uzi18, 2022-08-31:
> Ok, thank you! We should to discuss this idea with main developer.

@dukess they are compatible just change it and try

dukess, 2022-08-31:
@uzi18 thanks, we use OneWireNg since March. :)