PaulStoffregen/OneWire #111: esp32c3 compile error when using OneWire

ZJDATY, 2022-04-01:
Board
esp32c3

Device Description
D:\Users\XXX\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:226:18: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
GPIO.enable1_w1ts.val = ((uint32_t)1 << (pin - 32));
^~~~~~~~~~~~
enable_w1ts
exit status 1
为开发板 ESP32C3 Dev Module 编译时出错。

Hardware Configuration
gpio18 connect ds18b20 data

esp32 Version
v2.0.2

IDE Name
arduino

Operating System
windows 11

Flash frequency
40

PSRAM enabled
no

Upload speed
115200

Description
Compilation failed

Sketch
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

// arrays to hold device address
DeviceAddress insideThermometer;

/*
 * Setup function. Here we do the basics
 */
void setup(void)
{
  // start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature IC Control Library Demo");

  // locate devices on the bus
  Serial.print("Locating devices...");
  sensors.begin();
  Serial.print("Found ");
  Serial.print(sensors.getDeviceCount(), DEC);
  Serial.println(" devices.");

  // report parasite power requirements
  Serial.print("Parasite power is: "); 
  if (sensors.isParasitePowerMode()) Serial.println("ON");
  else Serial.println("OFF");
  
  // Assign address manually. The addresses below will beed to be changed
  // to valid device addresses on your bus. Device address can be retrieved
  // by using either oneWire.search(deviceAddress) or individually via
  // sensors.getAddress(deviceAddress, index)
  // Note that you will need to use your specific address here
  //insideThermometer = { 0x28, 0x1D, 0x39, 0x31, 0x2, 0x0, 0x0, 0xF0 };

  // Method 1:
  // Search for devices on the bus and assign based on an index. Ideally,
  // you would do this to initially discover addresses on the bus and then 
  // use those addresses and manually assign them (see above) once you know 
  // the devices on your bus (and assuming they don't change).
  if (!sensors.getAddress(insideThermometer, 0)) Serial.println("Unable to find address for Device 0"); 
  
  // method 2: search()
  // search() looks for the next device. Returns 1 if a new address has been
  // returned. A zero might mean that the bus is shorted, there are no devices, 
  // or you have already retrieved all of them. It might be a good idea to 
  // check the CRC to make sure you didn't get garbage. The order is 
  // deterministic. You will always get the same devices in the same order
  //
  // Must be called before search()
  //oneWire.reset_search();
  // assigns the first address found to insideThermometer
  //if (!oneWire.search(insideThermometer)) Serial.println("Unable to find address for insideThermometer");

  // show the addresses we found on the bus
  Serial.print("Device 0 Address: ");
  printAddress(insideThermometer);
  Serial.println();

  // set the resolution to 9 bit (Each Dallas/Maxim device is capable of several different resolutions)
  sensors.setResolution(insideThermometer, 9);
 
  Serial.print("Device 0 Resolution: ");
  Serial.print(sensors.getResolution(insideThermometer), DEC); 
  Serial.println();
}

// function to print the temperature for a device
void printTemperature(DeviceAddress deviceAddress)
{
  // method 1 - slower
  //Serial.print("Temp C: ");
  //Serial.print(sensors.getTempC(deviceAddress));
  //Serial.print(" Temp F: ");
  //Serial.print(sensors.getTempF(deviceAddress)); // Makes a second call to getTempC and then converts to Fahrenheit

  // method 2 - faster
  float tempC = sensors.getTempC(deviceAddress);
  if(tempC == DEVICE_DISCONNECTED_C) 
  {
    Serial.println("Error: Could not read temperature data");
    return;
  }
  Serial.print("Temp C: ");
  Serial.print(tempC);
  Serial.print(" Temp F: ");
  Serial.println(DallasTemperature::toFahrenheit(tempC)); // Converts tempC to Fahrenheit
}
/*
 * Main function. It will request the tempC from the sensors and display on Serial.
 */
void loop(void)
{ 
  // call sensors.requestTemperatures() to issue a global temperature 
  // request to all devices on the bus
  Serial.print("Requesting temperatures...");
  sensors.requestTemperatures(); // Send the command to get temperatures
  Serial.println("DONE");
  
  // It responds almost immediately. Let's print out the data
  printTemperature(insideThermometer); // Use a simple function to print out the data
}

// function to print a device address
void printAddress(DeviceAddress deviceAddress)
{
  for (uint8_t i = 0; i < 8; i++)
  {
    if (deviceAddress[i] < 16) Serial.print("0");
    Serial.print(deviceAddress[i], HEX);
  }
}
Debug Message
D:\Users\XXX\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:226:18: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
             GPIO.enable1_w1ts.val = ((uint32_t)1 << (pin - 32));
                  ^~~~~~~~~~~~
                  enable_w1ts
exit status 1
为开发板 ESP32C3 Dev Module 编译时出错。

uzi18, 2022-04-01:
@pstolarz are you ready with OneWireNg?

pstolarz, 2022-04-01:
C3 works fine on my lib. Future release 0.11 will support S3 and H2.

Jason2866, 2022-05-23:
PR #114

PaulStoffregen, 2022-05-23:
I've merged #114.  For everyone following this issue and still using ESP32, please download the latest OneWire and reply here to confirm (or deny) this fully solves the problem.

I am waiting for your feedback before publishing a new OneWire version which will make this fix available to everyone using the Arduino Library Manager.  Please take a moment to test this latest code and let me know if it should be published as a release to the wider Arduino community?

lpatiny, 2022-06-01:
I'm playing with ESP-32 (https://github.com/Hackuarium/esp32-c3/blob/main/src/taskOneWire.cpp) with various sensors and I can confirm that the head is working correctly.
Many thanks !

PaulStoffregen, 2022-06-04:
released 2.3.7

closed 2022-06-04 by PaulStoffregen