PaulStoffregen/OneWire #112: ESP32 compile error with arduino-esp32 v2.0.3-rc1 (error: 'esp32_gpioMux' was not declared in this scope)

bghavami, 2022-04-11:
**Board**
Esp32S2 Dev Module and Esp32S3 Dev Module

**Device Description**
NodeMcu-32S or Esp32-S3-DevKitC-1 v1.6
Testing with Dallas OneWire Temperature sensor

**Hardware Configuration**
GPIO 2 connected to Dallas Temperature sensor

**Version**
latest master

**IDE Name**
Arduino IDE 1.8.20

**Operating System**
Windows 10

**Flash frequency**
80 Mhz

**PSRAM enabled**
no

**Upload speed**
921600

**Description:**
I am getting an error while compiling sample/example code for Dallas Temperature sensor.
Steps to generate the error:
1- Load the code: File -> Examples -> Dallas Temperature -> Simple
2- verify / Compile

Compiler generates the following error:
OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;

It appears that "esp32_gpioMux" is no longer defined!!!

**Sketch:**
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

/*
 * The setup function. We only start the sensors here
 */
void setup(void)
{
  // start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature IC Control Library Demo");

  // Start up the library
  sensors.begin();
}

/*
 * Main function, get and show the temperature
 */
void loop(void)
{ 
  // call sensors.requestTemperatures() to issue a global temperature 
  // request to all devices on the bus
  Serial.print("Requesting temperatures...");
  sensors.requestTemperatures(); // Send the command to get temperatures
  Serial.println("DONE");
  // After we got the temperatures, we can print them here.
  // We use the function ByIndex, and as an example get the temperature from the first sensor only.
  float tempC = sensors.getTempCByIndex(0);

  // Check if reading was successful
  if(tempC != DEVICE_DISCONNECTED_C) 
  {
    Serial.print("Temperature for the device 1 (index 0) is: ");
    Serial.println(tempC);
  } 
  else
  {
    Serial.println("Error: Could not read temperature data");
  }
}

### Errors or Incorrect Output

Debug Message
In file included from C:\Users\eghav\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/Arduino.h:36,
                 from C:\Users\eghav\Documents\Arduino\libraries\OneWire\OneWire.cpp:142:
C:\Users\eghav\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
C:\Users\eghav\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope
         ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
                                      ^~~~~~~~~~~~~
C:\Users\eghav\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~
C:\Users\eghav\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
C:\Users\eghav\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:232:38: error: 'esp32_gpioMux' was not declared in this scope
         ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[pin].reg) = pinFunction;
                                      ^~~~~~~~~~~~~
C:\Users\eghav\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~
exit status 1
Error compiling for board ESP32S3 Dev Module.

vangalvin, 2022-04-15:
I seem to be having the same issue. I tried to move to OneWireNG but DSTherm is pretty limited in what it can do and i could not get the DallasTemperature lib to work with it.

uzi18, 2022-04-15:
OneWireNG and DallasTemperature works perfectly, what was your problem? If you got problem open issue in OneWireNG project
@pstolarz

vangalvin, 2022-04-15:
> OneWireNG and DallasTemperature works perfectly, what was your problem? If you got problem open issue in OneWireNG project @pstolarz

Its working, perfectly now. The issue was being caused by another version of onewire that was lurking in my libraries. As soon as I huned it out and deleted it the error went away.

uzi18, 2022-04-15:
We tested it on esp32 and it works with perfect timings on logic analyzer.
For esp32 it is perfect from 0.11.0

goetmat, 2022-04-19:
same issue on my side: `.pio/libdeps/xxx/OneWire/util/OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope`

bghavami, 2022-05-06:
> We tested it on esp32 and it works with perfect timings on logic analyzer. For esp32 it is perfect from 0.11.0 Glad you found solution.

If possible, Please, verify that it works with Esp32s3 Dev Module using Version 2.0.3-RC1. I am unable to get it to function.
Thanks.

uzi18, 2022-05-06:
If you got problem report it on OneWireNg project.
Thanks

juergen-weber, 2022-05-16:
I got a similar error on Platformio indicating that OneWire 2.3.6 is broken when being used with esp32-203:
[env:m5stack-core2]
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.3/esp32-2.0.3.zip
platform = espressif32
board = m5stack-core2

In function 'void directModeInput(uint32_t)':
.pio/libdeps/m5stack-core2/OneWire/util/OneWire_direct_gpio.h:191:38: error: 'esp32_gpioMux' was not declared in this scope

Dependency Graph
|-- <M5Core2> 0.1.2
|   |-- <Wire> 2.0.0
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|   |-- <HTTPClient> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |   |-- <WiFiClientSecure> 2.0.0
|   |   |   |-- <WiFi> 2.0.0
|   |-- <SD> 2.0.0
|   |   |-- <FS> 2.0.0
|   |   |-- <SPI> 2.0.0
|-- <DallasTemperature> 3.9.1
|   |-- <OneWire> 2.3.6
|-- <FS> 2.0.0
|-- <HTTPClient> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <WiFiClientSecure> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <SD> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <SPI> 2.0.0
|-- <SPIFFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <WiFi> 2.0.0
|-- <Wire> 2.0.0

Jason2866, 2022-05-16:
To be clear it is not a Arduino Core 2.0.3 issue. The problem is **external** OneWire library. The library is not from espressif. So espressif is not responsible for.

drquark, 2022-05-18:
For the previous comment: This was suggested on the Arduino Forum--
My only 2 suggestions would be to go to the github site of the onewire and report your issue there as you've done here. See if an older version of onewire will work.

ESPRESSIF has recently released a new version of the ESP32 core for the ESP-IDF. That core is not compatible, yet, with the ESP32 Arduino Core. From the errors OneWire is using the ESP32 API directly. OneWire may need to be updated.

uzi18, 2022-05-18:
But OneWireNG is compatible and works perfectly.

MrYsLab, 2022-05-18:
+1 I, too, am having the compile issue. Downgrading to version 2.0.2 of the ESP32 Arduino core works as a temporary fix.

drquark, 2022-05-18:
uzi18, I have read the docs on OneWireNG at github and I can't see how to do the following very simple read/write on a designated pin:
--------- for function "dallas()" --------------------------
int dallas(int x) {
OneWire ds(x) ;
int data0 ;
int data1 ;
int result ;

	ds.reset() ;
	ds.write(0xCC) ;  // skip command
	ds.write(0xBE) ;  // Read 1st 2 bytes of Scratchpad
	data0 = ds.read() ;
	data1 = ds.read() ;
	result = (data1 << 8) | data0 ;
	result = (((result * 90)+0x04) >> 3 ) + 3200 ;  // F x 100
	ds.reset() ;
	ds.write(0xCC) ;   // skip command
	ds.write(0x44,1) ; // start conversion
	return result ;
}

drquark, 2022-05-18:
For the previous comment, that's one DS18B20 per pin and direct power. Real simple.

What I don't see in the OneWireNG examples is how it is compatible with OneWire. If it is directly compatible, then the code in the previous comment should work without change. Is that true?

The reason I wrote my own code to read the DS18B20 is that I have 20 remote units, but the Dallas library can take a second or so to do a read. I don't have the time to sit in the Dallas function 20 times reading the remotes, so my code (the first time through in each remote) gets garbage, but it starts the conversion. The second and subsequent passes, it reads good data and restarts the conversion process. This works really well with a Nano or Pro Mini.

drquark, 2022-05-18:
OK, I used the library manager to add OneWireNg. The #include statement seems to work, but using the statement "OneWire" doesn't create the expected read/write functions, so I don't know what "compatible" means.

Jason2866, 2022-05-18:
@drquark  Use [this OneWire](https://github.com/arendst/Tasmota/tree/development/lib/lib_basic/OneWire-Stickbreaker) version. It is a modified version of OneWire and does work with actual Arduino Esp32 core 2.0.3

drquark, 2022-05-18:
OK, Jason2866, I'll give it a try. Thanks.

I just downgraded to ESP32 core version 2.0.2 and that eliminated the #include <OneWire.h> error.

uzi18, 2022-05-18:
> uzi18, I have read the docs on OneWireNG at github and I can't see how to do the following very simple read/write on a designated pin: --------- for function "dallas()" -------------------------- [...]

include is the same, but try to uninstall OneWire lib - they can collide in Arduino IDE

PaulStoffregen, 2022-05-18:
Please understand I am depending on the open source community to submit pull requests to maintain ESP32 defines.  
I do not personally use or test ESP boards.  The old defines, which worked with prior versions of the ESP software, were contributed by the open source community.  My hope is someone in the ESP open source community will take the time to verify a new set of defines and send a pull request.

drquark, 2022-05-18:
Using the ESP32 core version 2.0.2 and the current 2.3.6 version of OneWire, I'm uploading fine and reading the DS18B20s no sweat.

uzi18, 2022-05-18:
But now you are fixed with old core

Jason2866, 2022-05-18:
Have done a [version](https://github.com/Jason2866/OneWire) which is PR ready and should work. Since i have no possibility to test (for the next 4 weeks) give it a try and if ok please give me a feedback here and i will do the PR.

MrYsLab, 2022-05-18:
@Jason2866 Thanks for this fix. I just tried with a DS18B20 and am seeing some strange behavior.
If I test with ESP32 core version 2.0.2, your version works. However, if I upgrade to 2.0.3, my sketch compiles, but I cannot upload it to my board. There is a serial communication problem. I am using a DoIt DevKit1. 
Here is the failure output from the Arduino IDE with Show verbose output during compilation turned on:
```
Sketch uses 705725 bytes (53%) of program storage space. Maximum is 1310720 bytes.
Global variables use 42228 bytes (12%) of dynamic memory, leaving 285452 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.3-dev
Serial port /dev/ttyUSB0
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
the selected serial port For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
 does not exist or your board is not connected

```
I will research about the boot mode, and if I make any progress will report back.

juergen-weber, 2022-05-19:
Sorted. By simply replacing \.pio\libdeps\m5stack-core2\OneWire\utilOneWire_direct_gpio.h with your new file from https://github.com/Jason2866/OneWire/tree/esp32_arduino_core2/util/utilOneWire_direct_gpio.h solved the problem. Thanks

ogiewon, 2022-05-22:
@Jason2866 - Thank you very much!  I still need to perform some final testing, but so far my code now compiles and loads on to my ESP32-WROOM DevKit.

Jason2866, 2022-05-22:
> @Jason2866 Thanks for this fix. I just tried with a DS18B20 and am seeing some strange behavior. If I test with ESP32 core version 2.0.2, your version works. However, if I upgrade to 2.0.3, my [...]

This is issue is not related to the OneWire problem. You have a other problem here.

MrYsLab, 2022-05-23:
It works for me connecting to a DS18B20. Thanks.

ogiewon, 2022-05-26:
I just tested the latest version on an ESP8266 with 4 DS18B20 sensors attached to a single pin and it still works fine.  (I know this change is for the ESP32, but always good to make sure other platforms are not impacted as well.)

roboticboyer, 2022-05-28:
> I've merged #114. For everyone following this issue and still using ESP32, please download the latest OneWire and reply here to confirm (or deny) this fully solves the problem.
> 
> I am waiting for your feedback before publishing a new OneWire version which will make this fix available to everyone using the Arduino Library Manager. Please take a moment to test this latest code and let me know if it should be published as a release to the wider Arduino community?

It's working on ESP32!

MrYsLab, 2022-06-02:
Is there an ETA on when a new release containing this PR might be? I want to publish [my repository](https://github.com/MrYsLab/Telemetrix4Esp32) that depends on the fix or if I need to add some documentation to describe a workaround. Tnx.

PaulStoffregen, 2022-06-04:
2.3.7 released

closed 2022-06-04 by PaulStoffregen

MrYsLab, 2022-06-04:
@PaulStoffregen @Jason2866 Thank you.