PaulStoffregen/OneWire #116: volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg

Serega88kos, 2022-07-26:
Hi all. Tried various boards. All versions of the library. 

Logs Arduino IDE
..\libraries\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::reset()':
..\libraries\OneWire\OneWire.cpp:167:24: error: unused variable 'reg' [-Werror=unused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~
..\libraries\OneWire\OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':
..\libraries\OneWire\OneWire.cpp:201:24: error: unused variable 'reg' [-Werror=unused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~
..\libraries\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':
..\libraries\OneWire\OneWire.cpp:229:24: error: unused variable 'reg' [-Werror=unused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^~~
cc1plus.exe: some warnings being treated as errors
exit status 1
Compilation error for board DOIT ESP32 DEVKIT V1.

#include <OneWire.h>
#include "DallasTemperature.h"
#define ONE_WIRE_BUS 15
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature ds(&oneWire);

void setup() {
  Serial.begin(115200);
  ds.begin();
}
void loop() {
  ds.requestTemperatures();
  float temp = ds.getTempCByIndex(0);
  Serial.println(temp);
}

PaulStoffregen, 2022-07-26:
Does it work if you delete that line?  Or compile without -Werror?

Serega88kos, 2022-07-26:
//	volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
commented out three lines.
compilation completed.

closed 2022-07-26 by PaulStoffregen