PaulStoffregen/OneWire #123: rp2040 earlephilhower library compatibility

joyfab, 2022-12-06:
### Description

RP2040 (Raspberry pi pico) earlephilhower library) compatibility problem:
warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture. 
Right!. No way to run. Don't wire any gpio. Run only on mbedRP2040 boards.

### Steps Fix the Problem

open OneWire_direct_gpio.h.
just write line 445 at place of #define PIN_TO_BASEREG(pin)             (0)
by                                              #define PIN_TO_BASEREG(pin)             (pin)
comment boths warning (line 442 and 455 (//#warning ("OneWire. Fallback mode...
it's work like a charm on any gpio of the Pico, Pico-Zero, and Pico W
with earlephilhower library on Arduino IDE 1.8.19...;)

### Hardware & Software

Raspberry pi pico, W, and Zero.
Arduino IDE version 1.8.19 with earlephilhower library.
Version 2.6.3  https://github.com/earlephilhower/arduino-pico
Operating system
MacOS Sierra

use at the top #include <Arduino.h> and #include <DallasTemperature.h> for the sketch.
you can use any library from Arduino, it run with..

brentru, 2022-12-09:
@joyfab Could you explain more what you did? 

Making the change:
```
//#define PIN_TO_BASEREG(pin)             (0)
#define PIN_TO_BASEREG(pin) (pin)
```

results in the following error:
```
In file included from /Users/brentrubell/Documents/Arduino/libraries/OneWire/OneWire.cpp:144:
/Users/brentrubell/Documents/Arduino/libraries/OneWire/OneWire.cpp: In member function 'void OneWire::begin(uint8_t)':
/Users//Documents/Arduino/libraries/OneWire/OneWire.cpp:151:27: error: invalid conversion from 'uint8_t' {aka 'unsigned char'} to 'volatile unsigned int*' [-fpermissive]
  151 |  baseReg = PIN_TO_BASEREG(pin);
/Users//Documents/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:432:30: note: in definition of macro 'PIN_TO_BASEREG'
  432 | #define PIN_TO_BASEREG(pin) (pin)```

brentru, 2022-12-09:
The https://github.com/pstolarz/OneWireNg library does support RP2040 platforms, you may want to try that instead.

joyfab, 2022-12-10:
Yes! your right just 25 days written. I try it and feedback the result.
Problem was only to use non mbedrp2040 for this alternative onewire library. cannot use arduino.h for pin definitions. never worked except embed
Cool to transmit information, that open a way to use onewire on pico picoz and pico w with earlephilhower library !! ....;)
All with the great Arduino IDE and a large library easy to use. It's nice for each one who dont work on python or c++..

Thanks a lot for your work, am not really developper just using different library and test..

this is the modified OneWire_direct_gpio.h and for me it's really working dont know why, no problem.
Sure, correct with a good library it' really better.
you can take off my issue off and specify this location library for pico lowers...;) have a good day!

This is only what I changed. no error and good dallas temperature why any gpio on the rp2040.
correction is only if no board end of is detected by default. Arduino IDE 1.8.19 with earlephilhower library
include arduino.h.

here it's the really short working sketch.

joy.

joyfab, 2022-12-10:
So nice. tested . really cool to use on any rp2040! Good job, nice beerware! have a fresh one! top guy!
joy

witzed1, 2023-02-16:
Thanks for this information.  I'm almost a novice so can  you tell me once I open and edit the Onewire_direct_gpio.h file, How I then call it in a sketch.  There is no install option listed for this library in Arduino 2.0.3.  Thanks

chandradharkoneti, 2023-05-29:
I am having issues too. My PICO is completely becoming un-usable until I flash Nuke it.

I have tried the following boards and libraries.
*Boards:*
  - Raspberry Pi Pico/RP2040 (Earle F. Philhower, III) - v3.2.1
  - Arduino MBED OS RP2040 Boards (Arduino) - v4.0.2

*Libraries:*
  - OneWire - v2.3.5 and v2.3.7
  - DallasTemperature - v3.8.0 and v3.9.0
  - Arduino_Unified_Sensor - v1.1.9

None of them work. Please read further for output logs and windows error when my PICO bricks.

_I have attached the code in a zip file._
This is compilation and upload warning(Earle_Philhower board): Raspberry Pi Pico/RP2040 (Earle F. Philhower, III) - v3.2.1 ; OneWire - v2.3.7
```
In file included from D:\Dropbox\ARDUINO\INO files\libraries\OneWire\OneWire.cpp:144:
D:\Dropbox\ARDUINO\INO files\libraries\OneWire\util/OneWire_direct_gpio.h:441:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
  441 | #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
      |  ^~~~~~~
Sketch uses 55548 bytes (2%) of program storage space. Maximum is 2093056 bytes.
Global variables use 9056 bytes (3%) of dynamic memory, leaving 253088 bytes for local variables. Maximum is 262144 bytes.
Resetting COM15
Converting to uf2, output size: 144384, start address: 0x2000
Scanning for RP2040 devices
Flashing F: (RPI-RP2)
Wrote 144384 bytes to F:/NEW.UF2
```

A similar message is being displayed for Arduino MBED OS RP2040 Boards (Arduino) - v4.0.2.

Arduino MBED OS RP2040 Boards (Arduino) - v4.0.2 ; OneWire - v2.3.5
```
In file included from D:\Dropbox\ARDUINO\INO files\libraries\OneWire\OneWire.cpp:144:
D:\Dropbox\ARDUINO\INO files\libraries\OneWire\util/OneWire_direct_gpio.h:416:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
  416 | #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
      |  ^~~~~~~
Sketch uses 55412 bytes (2%) of program storage space. Maximum is 2093056 bytes.
Global variables use 9052 bytes (3%) of dynamic memory, leaving 253092 bytes for local variables. Maximum is 262144 bytes.
```

I have tried with Earle F Philhower's board as well as Arduino Original: MBED_OS_RP2040 board.

Both of them are causing my board to brick. Windows fails to recognise the device.
[image]

[NEW_DS18B20.zip](https://github.com/PaulStoffregen/OneWire/files/11590410/NEW_DS18B20.zip)

chandradharkoneti, 2023-05-29:
> ### Description
>
> RP2040 (Raspberry pi pico) earlephilhower library) compatibility problem: warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of [...]

I have tried the above suggestion, but It is failing to compile. 
```
In file included from D:\Dropbox\ARDUINO\INO files\libraries\OneWire\OneWire.cpp:144:
D:\Dropbox\ARDUINO\INO files\libraries\OneWire\OneWire.cpp: In member function 'void OneWire::begin(uint8_t)':
D:\Dropbox\ARDUINO\INO files\libraries\OneWire\OneWire.cpp:151:27: error: invalid conversion from 'uint8_t' {aka 'unsigned char'} to 'volatile unsigned int*' [-fpermissive]
  151 |  baseReg = PIN_TO_BASEREG(pin);
D:\Dropbox\ARDUINO\INO files\libraries\OneWire\util/OneWire_direct_gpio.h:431:42: note: in definition of macro 'PIN_TO_BASEREG'
  431 | #define PIN_TO_BASEREG(pin)             (pin)
      |                                          ^~~
Multiple libraries were found for "OneWire.h"
 Used: D:\Dropbox\ARDUINO\INO
 Not used: D:\Dropbox\ARDUINO\INO
Multiple libraries were found for "DallasTemperature.h"
 Used: D:\Dropbox\ARDUINO\INO
Multiple libraries were found for "Adafruit_Sensor.h"
 Used: D:\Dropbox\ARDUINO\INO
Multiple libraries were found for "DHT.h"
 Used: D:\Dropbox\ARDUINO\INO
exit status 1
Error compiling for board Raspberry Pi Pico.
```

I have searched the folder it mentioned and any other folders where libraries may be stored, but there are no duplicates of the libraries.

chandradharkoneti, 2023-06-07:
I have manually searched for the OneWire library's duplicate. I haven't found it anywhere(Including in any of adafruit/other third party libraries).
Anyhow, I'll try uninstalling Adafruit_Sensor library and let you know if that works. 

But Adafruit library is needed for a lot of my codes.

PaulStoffregen, 2023-06-07:
Click File > Preferences and turn on verbose output during compilation.  This will cause Arduino IDE to show the full pathnames of all libraries used.  It will also show all the compiler commands (a lot of text) which have the full pathnames to files used.