PaulStoffregen/OneWire #124: Compatibility issue with AVR  DxCore Family

sandeepkumar0153, 2023-02-12:
### Description

OneWire wont work on the new AVR Dx Series chips using board variant from SpenceKonde.

modified \Libraries\OneWire\2.3.7\OneWire\util\OneWire_direct_gpio.h

added condition in case ARDUINO_avrdb is defiend to achieve fall back mode and added following code modification. 
this should work with DA series and other boards with proper condition in to fall back mode.

#if defined(ARDUINO_avrdb)
#define PIN_TO_BASEREG(pin)             (0)
#define PIN_TO_BITMASK(pin)             (pin)
#define IO_REG_TYPE uint8_t
#define IO_REG_BASE_ATTR
#define IO_REG_MASK_ATTR
#define DIRECT_READ(base, pin)          digitalRead(pin)
#define DIRECT_WRITE_LOW(base, pin)     digitalWrite(pin, LOW)
#define DIRECT_WRITE_HIGH(base, pin)    digitalWrite(pin, HIGH)
#define DIRECT_MODE_INPUT(base, pin)    pinMode(pin,INPUT)
#define DIRECT_MODE_OUTPUT(base, pin)   pinMode(pin,OUTPUT)
#warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."

#elif defined(__AVR__)

nurazur, 2023-10-10:
Fallback mode did not work for me, when F_CPU clock is 4 MHz or below.
I am working on ATmega4808 and AVR dd devices. 
this line works, replacing line #18 in OneWire\util\OneWire_direct_gpio.h:
`#if defined(__AVR_ATmega4809__) ||  defined(__AVR_ATmega4808__) || defined (ARDUINO_avrdd)`
Since the architecture of ATmega480x and AVR Dx is very similar, I assume this to work for AVR db series as well if you add
` || defined (ARDUINO_avedb) ` to line #18

nurazur, 2024-01-21:
typo in my previous post is corrected. Sorry for the inconvenience.

WeirdConstructor, 2024-08-17:
Also stumbled across this one today (had to add ARDUINO_avrdd to line 18).

paultheg33k, 2025-04-08:
Confirming this works for ATTiny 1624 chips if you likewise add : 

`|| defined(__AVR_ATtiny1624__) `

to line 18.