PaulStoffregen/OneWire #128: It is desirable to support new AVR controllers.

mikrocoder, 2023-06-21:
Hi Paul,

would you please include support for the new AVR controllers in your OneWire?
For everything concerning the AVR, only this small change is necessary.
Tested with AVR128DB64, Nano Every (ATmega4809) and Arduino Mega2560.
You can use the completely revised and better readable formatted file directly. I have labeled it v2.3.8 for better distinction.
[OneWire_direct_gpio_v2.3.8.zip](https://github.com/PaulStoffregen/OneWire/files/11823769/OneWire_direct_gpio_v2.3.8.zip)
It would be very kind of you.

PaulStoffregen, 2023-06-21:
This is too soon to ask again.  Wait until next week.

closed 2023-06-21 by PaulStoffregen

mikrocoder, 2023-06-21:
I thought I should close the old issue and I should open a new issue with clear wording?

PaulStoffregen, 2023-06-21:
Stop right now.  Do not write another word, no more replies today.  Wait until next week.