PaulStoffregen/OneWire #130: [Warning] unused variable `reg`

hasenradball, 2023-09-15:
Hi together,

I see the warnings when using the latest release.
Should these not be fixed actually?

[image]

----------------------------- Remove above -----------------------------

### Description

Describe your problem.

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

### Hardware & Software

Board:                             Wemos Mini D1
Shields / modules used:   No
Arduino IDE version:       2.2.1
Teensyduino version (if using Teensy)
Version info & package name : 3.0.2
Operating system & version
Any other software or hardware?

hasenradball, 2023-09-15:
Hi One addition,

on the ESP32 Pico D4
There are the warnings teated as erros...

[image]

hasenradball, 2023-10-02:
Hello all,

why do I get this error?

Can we mark these variables with attribute unused?

[image]

hasenradball, 2023-10-02:
@PaulStoffregen, @Jason2866 
Hi Paul,

I think with the release `2.3.7` there is something wrong!!!
I saw on github the changes for the Attribute `unused variable`, but when using the version 2.3.7 the changes are not inside the zip-file.

Also on PIO the changes are not inside.

Maybee you have to check the release plese.

Thanks

Frank

hasenradball, 2023-11-30:
@PaulStoffregen 
Can you please publish your changes again on PIO.
Thanks.

PaulStoffregen, 2023-12-11:
Sorry, I do now know how to check PIO.

PaulStoffregen, 2023-12-11:
Can you try using with Arduino IDE and confirm if the problem exists with Arduino library manager?

hasenradball, 2023-12-11:
@PaulStoffregen 
Hi Paul,
it seems also on the Arduino library it is the same issue.
It seems the changes are not in the release 2.3.7.

Jason2866, 2023-12-11:
Probably it makes sense to add the needed changes for upcoming espressif32 Arduino Core 3.0.0 before doing a new release.

PaulStoffregen, 2023-12-11:
I've merged several ESP related pull requests.  But I didn't see any for espressif32 Arduino Core 3.0.0.  If there is a PR pending, can you tell me which one I missed?