PaulStoffregen/OneWire #131: Cannot use ESP32 S3 GPIO 43, 44 for OneWire Dallas DS18B20 due to setting in util file

teastainGit, 2023-09-20:
Attempting to run a DS18B20 Dallas Semiconductor temp sensor on the "I2C" JST connector of a LilyGO T-Display S3.
The reading was -127 when I tried either pin 43 or pin 44.
Pins 1, 2, 18 or 17 worked fine.
Sensor would not work on 43, 44 on a M5StampS3, but would on any other.
Contributors to the Espressif Arduino repository 
https://github.com/espressif/arduino-esp32/issues/8324#event-10412569686
finally after 4 months hit on the possibility of the library's Util file OneWire_direct_gpio.h line 200; limiting the pin count to 33, the remainders would be "inputs".
I edited the util header and it worked.
Should I make a Pull Request?
-Terry

Jason2866, 2023-10-02:
Imho a new release should include the needed changes for the upcoming espressif32 Arduino 3.0. We use a modified forked version and did the changes. Since it is not based on the current version, the changes done needs to be ported to the version here.
The modified working for core 2.0.x and 3.0.x (support for 1.0.x is dropped) is here
https://github.com/arendst/Tasmota/tree/development/lib/lib_basic/NeoPixelBus
Anyone feel free to do the PR

369Martin369, 2024-01-20:
> Arduino IDE 2.2.1 all libraries updated to current available. Attempting to run a DS18B20 Dallas Semiconductor temp sensor on the "I2C" JST connector of a LilyGO T-Display S3. The reading was -127 when I tried either pin 43 or pin 44. Pins 1, 2, 18 or 17 worked fine. Sensor would not work on 43, 44 on a M5StampS3, but would on any other. Contributors to the Espressif Arduino repository [espressif/arduino-esp32#8324 (comment)](https://github.com/espressif/arduino-esp32/issues/8324#event-10412569686) finally after 4 months hit on the possibility of the library's Util file OneWire_direct_gpio.h line 200; limiting the pin count to 33, the remainders would be "inputs". I edited the util header and it worked. Should I make a Pull Request? -Terry

What exactly did you edit? Line 200 set to 44 does not work...?!? help please

teastainGit, 2024-01-20:
Martin:
Well, I just set it to 44 as a local, ad-hoc edit, as the simplest solution, but that line NOW in the github  repository  says:
``` if ( digitalPinIsValid(pin) && pin <= NUM_DIGITAL_PINS ) // pins above 33 can be only inputs```
My libraries are up-to-date and the sensor works on pin44, as I write.
Hope this helps.
-Terry

369Martin369, 2024-01-20:
Thanks. meanwhile I used the patched version Tasmota https://github.com/arendst/Tasmota/tree/development/lib/lib_basic/OneWire-Stickbreaker 
and it works now :-) Thanks

patofoto, 2024-06-14:
Have tried everything in this thread and also this:

[https://github.com/PaulStoffregen/OneWire/issues/141](https://github.com/PaulStoffregen/OneWire/issues/141)

No luck yet. Is there any other library that might allow me to read PIN 43?. I need to use the JST connector.

uzi18, 2024-06-14:
@patofoto this one supports pins >32 https://github.com/pstolarz/OneWireNg