PaulStoffregen/OneWire #132: Arduino IDE 2.1 with ESP32 board 3.0.0-alpha2 : OneWire not compatible

sbarabe, 2023-11-19:
I was trying to use OneWire with ESP32 bards and had error at compilation about GPIO.in.val not existing... I've rolllback to ESP32 2.0.14 and it compile just fine.

So you should expect request about this sooner or later.

I had the same issue with the TFT_eSPI library but they emitted a update version and it's fine now.

Thanks for the library !

YCottreau, 2023-11-24:
Hi,
I have the same issue using ESP32 3.0.0 with OneWire library.
Thanks if you solve this soon, the 3.0.0 official release is annonced for end of decembre :-)

Emilevanderlaan, 2024-01-07:
I have the same problem , I use the ESP32S3 and ONEWIRE_PIN  4

In file included from c:\Users\emile\OneDrive\Documents\Arduino\libraries\OneWire\OneWire.cpp:144:
c:\Users\emile\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'uint32_t directRead(uint32_t)':
c:\Users\.\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:134:17: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  134 |         return (GPIO.in >> pin) & 0x1;
      |                 ^~~~
      |                 PI
c:\Users\.\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:136:17: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  136 |         return (GPIO.in1.val >> (pin - 32)) & 0x1;
      |                 ^~~~
      |                 PI
c:\Users\..\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directWriteLow(uint32_t)':
c:\Users\..\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:149:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  149 |         GPIO.out_w1tc = ((uint32_t)1 << pin);
      |         ^~~~
      |         PI
c:\Users\.\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:151:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  151 |         GPIO.out1_w1tc.val = ((uint32_t)1 << (pin - 32));
      |         ^~~~
      |         PI
c:\Users\..\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directWriteHigh(uint32_t)':
c:\Users\..\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:162:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  162 |         GPIO.out_w1ts = ((uint32_t)1 << pin);
      |         ^~~~
      |         PI
c:\Users\..\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:164:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  164 |         GPIO.out1_w1ts.val = ((uint32_t)1 << (pin - 32));
      |         ^~~~
      |         PI
c:\Users\.\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
c:\Users\..\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:187:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  187 |             GPIO.enable_w1tc = ((uint32_t)1 << pin);
      |             ^~~~
      |             PI
c:\Users\..\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:189:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  189 |             GPIO.enable1_w1tc.val = ((uint32_t)1 << (pin - 32));
      |             ^~~~
      |             PI
c:\Users\.\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
c:\Users\.\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:213:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  213 |             GPIO.enable_w1ts = ((uint32_t)1 << pin);
      |             ^~~~
      |             PI
c:\Users\..\OneDrive\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:215:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  215 |             GPIO.enable1_w1ts.val = ((uint32_t)1 << (pin - 32));
      |             ^~~~
      |             PI

seansplayin, 2024-01-27:
Same error, rolled back esp32 3.0.0-alpha3 to 2.1.14

closed 2024-02-08 by PaulStoffregen

schreiekm, 2024-05-06:
Issue appearing again upwards from version 2.3.3 and adding `#include <soc/gpio_struct.h>` doesn't fix it.