PaulStoffregen/OneWire #133: Bug in release 2.3.8: extra tokens at end of #undef directive

pixelfriese, 2023-12-13:
Please use this form only to report code defects or bugs.

For any question, even questions directly pertaining to this code, post your question on the forums related to the board you are using.

Arduino: forum.arduino.cc
Teensy: forum.pjrc.com
ESP8266: www.esp8266.com
ESP32: www.esp32.com
Adafruit Feather/Metro/Trinket: forums.adafruit.com
Particle Photon: community.particle.io

If you are experiencing trouble but not certain of the cause, or need help using this code, ask on the appropriate forum.  This is not the place to ask for support or help, even directly related to this code.  Only use this form you are certain you have discovered a defect in this code!

Please verify the problem occurs when using the very latest version, using the newest version of Arduino and any other related software.

----------------------------- Remove above -----------------------------

### Description

Since the new version 2.3.8, error messages are displayed during the build process.

### Steps To Reproduce Problem

The error occurs directly in the build process of the library. Platformio is used with an ESP32 DevKit V1. OneWire is used as a dependency for the milesburton/DallasTemperature @ 3.11.0 library.

### Hardware & Software

Board ESP32 DevKitV1
Shields / modules used no
Arduino IDE version PlatformIO 6.1.11
Teensyduino version (if using Teensy) no
Version info & package name (from Tools > Boards > Board Manager)  paulstoffregen/OneWire @ 2.3.8
Operating system & version Windows 11
Any other software or hardware? no

### Arduino Sketch

OneWire\OneWire.cpp

```cpp
// undef defines for no particular reason
#ifdef ARDUINO_ARCH_ESP32
#  undef noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
#  undef interrupts() portEXIT_CRITICAL(&mux);}
#endif
// for info on this, search "IRAM_ATTR" at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/general-notes.html 
#undef CRIT_TIMING 
```

### Errors or Incorrect Output

```
.pio/libdeps/myproject/OneWire/OneWire.cpp:599:22: warning: extra tokens at end of #undef directive
 #  undef noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
                      ^
.pio/libdeps/myproject/OneWire/OneWire.cpp:600:20: warning: extra tokens at end of #undef directive
 #  undef interrupts() portEXIT_CRITICAL(&mux);}
```

Nik-mayak, 2024-01-22:
the problem is solved like this: in file OneWire.cpp
#ifdef ARDUINO_ARCH_ESP32
#  undef noInterrupts//() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
#  undef interrupts//() portEXIT_CRITICAL(&mux);}
#endif

LoQue90, 2024-02-28:
@Nik-mayak are you going to open a PR on this?

mathieucarbou, 2024-03-17:
@PaulStoffregen : would it be possible to merge [this PR](https://github.com/PaulStoffregen/OneWire/pull/135) and issue a new release and deploy in platformio registry ?
Thanks!