PaulStoffregen/OneWire #136: ESP32 version 3.0.0-alpha compiling error

Pr77Pr77, 2024-03-31:
### Description

The compiling of the OneWire libary with the pre-released ESP32 core version 3.0.0-alpha3 throws an errorwhile compiling the DS18x20_Temperature example.

### Steps To Reproduce Problem

Install the latest version of OneWire (2.3.7) and the ESP32 core version 3.0.0-alpha3. Choose the board "ESP32 Dev Module" and compile the example "DS18x20_Temperature".

### Hardware & Software

Board: ESP32 Dev Module
Shields / modules used: I don't know
Arduino IDE version: 2.3.2
Teensyduino version (if using Teensy): Not used
Version info & package name (from Tools > Boards > Board Manager): esp32 by Espressif Systems Version 3.0.0-alpha3
Operating system & version: Windows 11
Any other software or hardware?: No

### Arduino Sketch

DS18x20_Temperature example of OneWire:
```cpp
#include <OneWire.h>

// OneWire DS18S20, DS18B20, DS1822 Temperature Example
//
// http://www.pjrc.com/teensy/td_libs_OneWire.html
//
// The DallasTemperature library can do all this work for you!
// https://github.com/milesburton/Arduino-Temperature-Control-Library

OneWire  ds(10);  // on pin 10 (a 4.7K resistor is necessary)

void setup(void) {
  Serial.begin(9600);
}

void loop(void) {
  byte i;
  byte present = 0;
  byte type_s;
  byte data[9];
  byte addr[8];
  float celsius, fahrenheit;
  
  if ( !ds.search(addr)) {
    Serial.println("No more addresses.");
    Serial.println();
    ds.reset_search();
    delay(250);
    return;
  }
  
  Serial.print("ROM =");
  for( i = 0; i < 8; i++) {
    Serial.write(' ');
    Serial.print(addr[i], HEX);
  }

  if (OneWire::crc8(addr, 7) != addr[7]) {
      Serial.println("CRC is not valid!");
      return;
  }
  Serial.println();
 
  // the first ROM byte indicates which chip
  switch (addr[0]) {
    case 0x10:
      Serial.println("  Chip = DS18S20");  // or old DS1820
      type_s = 1;
      break;
    case 0x28:
      Serial.println("  Chip = DS18B20");
      type_s = 0;
      break;
    case 0x22:
      Serial.println("  Chip = DS1822");
      type_s = 0;
      break;
    default:
      Serial.println("Device is not a DS18x20 family device.");
      return;
  } 

  ds.reset();
  ds.select(addr);
  ds.write(0x44, 1);        // start conversion, with parasite power on at the end
  
  delay(1000);     // maybe 750ms is enough, maybe not
  // we might do a ds.depower() here, but the reset will take care of it.
  
  present = ds.reset();
  ds.select(addr);    
  ds.write(0xBE);         // Read Scratchpad

  Serial.print("  Data = ");
  Serial.print(present, HEX);
  Serial.print(" ");
  for ( i = 0; i < 9; i++) {           // we need 9 bytes
    data[i] = ds.read();
    Serial.print(data[i], HEX);
    Serial.print(" ");
  }
  Serial.print(" CRC=");
  Serial.print(OneWire::crc8(data, 8), HEX);
  Serial.println();

  // Convert the data to actual temperature
  // because the result is a 16 bit signed integer, it should
  // be stored to an "int16_t" type, which is always 16 bits
  // even when compiled on a 32 bit processor.
  int16_t raw = (data[1] << 8) | data[0];
  if (type_s) {
    raw = raw << 3; // 9 bit resolution default
    if (data[7] == 0x10) {
      // "count remain" gives full 12 bit resolution
      raw = (raw & 0xFFF0) + 12 - data[6];
    }
  } else {
    byte cfg = (data[4] & 0x60);
    // at lower res, the low bits are undefined, so let's zero them
    if (cfg == 0x00) raw = raw & ~7;  // 9 bit resolution, 93.75 ms
    else if (cfg == 0x20) raw = raw & ~3; // 10 bit res, 187.5 ms
    else if (cfg == 0x40) raw = raw & ~1; // 11 bit res, 375 ms
    //// default is 12 bit resolution, 750 ms conversion time
  }
  celsius = (float)raw / 16.0;
  fahrenheit = celsius * 1.8 + 32.0;
  Serial.print("  Temperature = ");
  Serial.print(celsius);
  Serial.print(" Celsius, ");
  Serial.print(fahrenheit);
  Serial.println(" Fahrenheit");
}
```

### Errors or Incorrect Output

In file included from [path to documents]\Arduino\libraries\OneWire\OneWire.cpp:144:
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'uint32_t directRead(uint32_t)':
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:134:17: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  134 |         return (GPIO.in >> pin) & 0x1;
      |                 ^~~~
      |                 PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:136:17: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  136 |         return (GPIO.in1.val >> (pin - 32)) & 0x1;
      |                 ^~~~
      |                 PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directWriteLow(uint32_t)':
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:149:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  149 |         GPIO.out_w1tc = ((uint32_t)1 << pin);
      |         ^~~~
      |         PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:151:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  151 |         GPIO.out1_w1tc.val = ((uint32_t)1 << (pin - 32));
      |         ^~~~
      |         PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directWriteHigh(uint32_t)':
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:162:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  162 |         GPIO.out_w1ts = ((uint32_t)1 << pin);
      |         ^~~~
      |         PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:164:9: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  164 |         GPIO.out1_w1ts.val = ((uint32_t)1 << (pin - 32));
      |         ^~~~
      |         PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:187:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  187 |             GPIO.enable_w1tc = ((uint32_t)1 << pin);
      |             ^~~~
      |             PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:189:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  189 |             GPIO.enable1_w1tc.val = ((uint32_t)1 << (pin - 32));
      |             ^~~~
      |             PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:213:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  213 |             GPIO.enable_w1ts = ((uint32_t)1 << pin);
      |             ^~~~
      |             PI
[path to documents]\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:215:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  215 |             GPIO.enable1_w1ts.val = ((uint32_t)1 << (pin - 32));
      |             ^~~~
      |             PI

code-reboot-F, 2024-05-28:
我也遇到了同样的问题（甚至报错显示的行数都一样XD），退回开发板的上一个版本（2.0.17）可以临时解决问题，但是我确实想在3.0.0中使用，我尝试修改库文件使得编译通过，不过失败了。。
目前我通过报错与库文件中寻找到可能与问题相关的开发板文件有：rtc_io.h和gpio.h，但是我去乐鑫官网找到的从2.0.17迁移到3.0.0的文档中似乎没有提及这部分
[https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html]

Pr77Pr77, 2024-05-28:
The problem still exists with version 3.0.0.

peff74, 2024-05-30:
Just use the new  [OneWire_direct_gpio.h](https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h)
You have to replace it with the old one in your library directory.

code-reboot-F, 2024-05-30:
> 版本 3.0.0 仍然存在问题。
看来是onewire的库需要更新，使用@peff74 的链接获取新的.h文件并替换旧文件之后可以编译通过

code-reboot-F, 2024-05-30:
> Just use the new [OneWire_direct_gpio.h](https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h) You have to replace it with the old one in your library directory.

万分感谢

per1234, 2024-05-31:
Duplicate of https://github.com/PaulStoffregen/OneWire/issues/132, which has already been fixed by https://github.com/PaulStoffregen/OneWire/pull/134.

@Pr77Pr77 please close this issue to keep the issue tracker tidy.

Pr77Pr77, 2024-05-31:
Replacing the [OneWire_direct_gpio.h](https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h) file works. Thank you.

closed 2024-05-31 by Pr77Pr77

PaulStoffregen, 2024-05-31:
I've tagged release 2.3.8.  It had been prepared some time ago, but was forgotten after lack of community feedback.

code-reboot-F, 2024-05-31:
> 我已经标记了版本 2.3.8。它早在一段时间前就准备好了，但由于缺乏社区反馈而被遗忘了。

😅😅😅😅😅😅😅😅😅