PaulStoffregen/OneWire #141: Support for Pin 48 on ESP32-S3 in OneWire Library

KaoruTK, 2024-05-20:
### Description
We encountered an issue with the OneWire library on the ESP32-S3 when using pin 48. The library originally only supported pins up to 36. To enable the use of pin 48, we modified the OneWire_direct_gpio.h file.

### Steps To Reproduce Problem

Use an ESP32-S3 board.
Connect a DS18B20 sensor to pin 48.
Attempt to read data using the OneWire library without the modifications.
Observe that it does not work due to pin limitations in the library.

### Hardware & Software

Board: ESP32-S3
Modules used: DS18B20 temperature sensor
Arduino IDE version: 1.8.19
ESP32 Board Manager package version: 2.3.2
Operating system & version: Windows 10
Other software/hardware: None

### Arduino Sketch

#include <OneWire.h>                
#include <DallasTemperature.h>
 
OneWire ourWire(48);                //Se establece el pin 2  como bus OneWire
 
DallasTemperature sensors(&ourWire); //Se declara una variable u objeto para nuestro sensor

void setup() {
delay(1000);
Serial.begin(9600);
sensors.begin();   //Se inicia el sensor
}
 
void loop() {
sensors.requestTemperatures();   //Se envía el comando para leer la temperatura
float temp= sensors.getTempCByIndex(0); //Se obtiene la temperatura en ºC

Serial.print("Temperatura= ");
Serial.print(temp);
Serial.println(" C");
delay(100);                     
}

### Errors or Incorrect Output

Without the modifications, the OneWire library does not recognize pin 48 as valid, causing the sensor to be undetected. The issue arises because the original library code only supports pins up to 36. After modification, the library should support higher pins, including pin 48, on the ESP32-S3.

### Code Modifications
We modified the following lines in the OneWire_direct_gpio.h file to support pins higher than 36:

**directRead function:**

static inline __attribute__((always_inline))
IO_REG_TYPE directRead(IO_REG_TYPE pin)
{
#if CONFIG_IDF_TARGET_ESP32C3
    return (GPIO.in.val >> pin) & 0x1;
#else // plain ESP32
    if (pin < 32)
        return (GPIO.in >> pin) & 0x1;
    else
        return (GPIO.in1.val >> (pin - 32)) & 0x1;
#endif

    return 0;
}
**directWriteLow function:**

static inline __attribute__((always_inline))
void directWriteLow(IO_REG_TYPE pin)
{
#if CONFIG_IDF_TARGET_ESP32C3
    GPIO.out_w1tc.val = ((uint32_t)1 << pin);
#else // plain ESP32
    if (pin < 32)
        GPIO.out_w1tc = ((uint32_t)1 << pin);
    else
        GPIO.out1_w1tc.val = ((uint32_t)1 << (pin - 32));
#endif
}
**directWriteHigh function:**

static inline __attribute__((always_inline))
void directWriteHigh(IO_REG_TYPE pin)
{
#if CONFIG_IDF_TARGET_ESP32C3
    GPIO.out_w1ts.val = ((uint32_t)1 << pin);
#else // plain ESP32
    if (pin < 32)
        GPIO.out_w1ts = ((uint32_t)1 << pin);
    else
        GPIO.out1_w1ts.val = ((uint32_t)1 << (pin - 32));
#endif
}
**directModeInput function:**

static inline __attribute__((always_inline))
void directModeInput(IO_REG_TYPE pin)
{
#if CONFIG_IDF_TARGET_ESP32C3
    GPIO.enable_w1tc.val = ((uint32_t)1 << (pin));
#else
    if (digitalPinIsValid(pin))
    {
#if ESP_IDF_VERSION_MAJOR < 4      // IDF 3.x ESP32/PICO-D4
        uint32_t rtc_reg(rtc_gpio_desc[pin].reg);

        if (rtc_reg) // RTC pins PULL settings
        {
            ESP_REG(rtc_reg) = ESP_REG(rtc_reg) & ~(rtc_gpio_desc[pin].mux);
            ESP_REG(rtc_reg) = ESP_REG(rtc_reg) & ~(rtc_gpio_desc[pin].pullup | rtc_gpio_desc[pin].pulldown);
        }
#endif
        // Input
        if (pin < 32)
            GPIO.enable_w1tc = ((uint32_t)1 << pin);
        else
            GPIO.enable1_w1tc.val = ((uint32_t)1 << (pin - 32));
    }
#endif
}
**directModeOutput function:**

static inline __attribute__((always_inline))
void directModeOutput(IO_REG_TYPE pin)
{
#if CONFIG_IDF_TARGET_ESP32C3
    GPIO.enable_w1ts.val = ((uint32_t)1 << (pin));
#else
    if (digitalPinIsValid(pin))
    {
#if ESP_IDF_VERSION_MAJOR < 4      // IDF 3.x ESP32/PICO-D4
        uint32_t rtc_reg(rtc_gpio_desc[pin].reg);

        if (rtc_reg) // RTC pins PULL settings
        {
            ESP_REG(rtc_reg) = ESP_REG(rtc_reg) & ~(rtc_gpio_desc[pin].mux);
            ESP_REG(rtc_reg) = ESP_REG(rtc_reg) & ~(rtc_gpio_desc[pin].pullup | rtc_gpio_desc[pin].pulldown);
        }
#endif
        // Output
        if (pin < 32)
            GPIO.enable_w1ts = ((uint32_t)1 << pin);
        else
            GPIO.enable1_w1ts.val = ((uint32_t)1 << (pin - 32));
    }
#endif
}
By making these modifications, the library can now work with pins higher than 36, including pin 48 on the ESP32-S3. This change should be reviewed and potentially integrated to support a wider range of pins in the ESP32-S3 series.

patofoto, 2024-06-14:
Hi there. I am trying to use pin 43 to read a Dallas Sensor and can't seem to make it work. I believe I have replaced the code in the `OneWire_direct_gpio.h` correctly but don't really know. Here is my file. Any help would be greatly appreciated.

[OneWire_direct_gpio.h.zip](https://github.com/user-attachments/files/15838284/OneWire_direct_gpio.h.zip)

uzi18, 2024-06-14:
this lib supports pins >32 https://github.com/pstolarz/OneWireNg

patofoto, 2024-06-14:
@uzi18 Thank you. Will look into it this weekend

DennisDG, 2024-07-31:
Works like a charm! thank you mate!

uzi18, 2024-07-31:
@DennisDG lib or patch?

vgoikhman, 2025-02-14:
Works now. Thank you!