PaulStoffregen/OneWire #142: DS1820 temperature readings ESP32 in combination with FreeRTOS tasks leads to malfunction of the RMT Module

sirpete83, 2024-05-25:
ESP32 Wemos Lolin32

sketch with 3 additional FreeRTOS tasks leads to a significant lengthening of the set poriodes of the rmt module. I was able to narrow it down to the Onewire or the Dalles temperature library. If I remove Onewire completely from the sketch, then the times are correct again.
which surprises me a bit, because the onewire library doesn't seem to use rmt. And it's just fed with data and then runs in hardware, at least that's my assumption. 
can someone confirm the problem?

sirpete83, 2024-05-26:
Hallo,
I looked at the OneWire code again and saw that this library deactivates the interrupts in order to maintain the timings. It's certainly not a problem for non-RTOS applications. But it will certainly stop all hardware timers and thus halt the tasks. 
I don't expect a solution to the problem, but perhaps this insight will help others if they encounter timing problems in their RTOs application when using this library.

tshcherban, 2024-06-06:
There is a [RMT-based](https://github.com/junkfix/esp32-ds18b20) OneWire library, so if you have a free RMT instance - u can use it for onewire communication

uzi18, 2024-06-08:
just try OneWireNg library